olofk/edalize #34: Setting strategy/flow for synthesis and implementation for Vivado.

m-kru, 2019-06-25:
I would like to be able to define strategy and flow for synthesis and implementation in .core files the same way as I can define part.
For Vivado backend tool_options is very simple and looks as follows:    
`tool_options = {'members' : {'part' : 'String'}}`
I have also checked files for other tools and for Quartus tools_options looks like this:
```
  tool_options = {'members' : {'family' : 'String',
                               'device' : 'String'},
                  'lists'   : {'quartus_options' : 'String'}}
```
I think I don't fully get the idea behind these `members` and `lists`. What is the difference between them? Should I add strategy and flow options to Vivado as members or maybe I should create lists, the same way as it is done for Quartus?

olofk, 2019-06-25:
I got two answers here
1. I prefer to keep options like this as tcl scripts as far as possible to avoid an explosion of backend options. So if you want to e.g. change the synthesis strategy, you can create a new file, write `set_property strategy Flow_AreaOptimized_high [get_runs synth_1]` in it and add it to your core file with `file_type : tclSource`.

2. If there is still a good reason to add it as a backend option (or if you need to add something else), things are a bit complicated right now. The `tool_options` struct is deprecated because of some...issues (long story), but it can't be removed yet. The (I admit not very obvious) way to add new options is to add it to the struct in the `get_doc` function where you will currently find `part` and `pnr`. `members` are when you want to add a single string as the option and `lists` is when you need to have a list of values. In the example you added above, you can only have one device and family, but quartus_options could be a list of different options

m-kru, 2019-06-26:
Option number 1 works as I expect it to work. As I agree with you that minimizing backend options is good I am closing this issue.

closed 2019-06-26 by m-kru