olofk/edalize #42: vivado-project.tcl.j2 not found

AnttiLukats, 2019-07-31:
blinky is not working, some files are generated then error comes with template not found exeption :(

olofk, 2019-08-01:
Thanks for reporting, and you know what? I managed to boot into Windows long enough to install and reproduce the issue. And not only that. I think I fixed it and released a new version, so please try edalize 0.1.6 and let me know if your problems go away.

AnttiLukats, 2019-08-02:
I see absolutely no difference exactly the same error

olofk, 2019-08-02:
Hmm.. are you sure edalize got upgraded? (`pip install -U edalize` otherwise). The reason I'm asking is that there was a backslash in `jinja2.exceptions.TemplateNotFound: vivado\vivado-project.tcl.j2` from the error log you reported. This should now be a regular slash after edalize 0.1.6 so I can't understand how it can be the exact same error message after the upgrade

AnttiLukats, 2019-08-02:
my bad, I tried all I could to update fusesoc and edalize with github client, but I guess it was nonsense, it is building now, thanks..

closed 2019-08-02 by AnttiLukats