olofk/edalize #43: Vivado backend 'file_type=data' type

djcopley, 2019-08-01:
In the vivado backend source code it says:
```
- IP: Supply the IP core xci file with file_type=xci and other files (like .prj)
      as file_type=data
```
However, this is the only reference to file_type=data that I found. Is this a mistake?

olofk, 2019-08-01:
Yes, this is a mistake. That should be user instead of data

djcopley, 2019-08-01:
How does realize treat user files? Does it just copy them to the ../src dir?

olofk, 2019-08-01:
Yes and no. Technically FuseSoC does the copying and edalize promises to not do anything at all. So the main use is to get files in the build tree without anyone touching them. This is for example used for data files that are read by HDL files or scripts during the build process

closed 2019-08-01 by djcopley