olofk/edalize #44: Vivado HLS support

qarlosalberto, 2019-08-01:
I think it might be interesting to integrate hlsclt ( https://github.com/benjmarshall/hlsclt )  to have support for Vivado HLS.

olofk, 2019-08-26:
That would be a good addition. Unfortunately I know too little about Vivado HLS. Wondering if it would be best done as a completely new edalize backend or if it would be better to have as a generator that just compiles the code to RTL and reuses the rest of the infrastructure after that

1138-4EB, 2019-08-26:
> Wondering if it would be best done as a completely new edalize backend or if it would be better to have as a generator that just compiles the code to RTL and reuses the rest of the infrastructure after that

I think that this depends on the project. It is possible to export HLS designs to RTL (VHDL) and then simulate them with GHDL + VUnit verification components. However, the software partition might need rewriting.