olofk/edalize #48: ISE backend support for general "project set" commands

GCHQDeveloper560, 2019-09-03:
Edalize currently uses ISE's Tcl interface to create a project and set project options via the `project set` group of commands. It would be helpful if a user could add their own settings in YAML, perhaps under `tool_options`. Some properties apply to multiple steps in the process and require the `-process` argument. For example:

```
tool_options:
    device: ...
    project_properties:
        "Allow Logic Optimization Across Hierarchy":
            value: false
        LUT Combining:
            value: true
            process: "Synthesize - XST"
```

ISE project file generation is currently done with an Python string formatting and file writes. Presumably step one would be to convert this to Jinja templating like several other backends. After that this addition should be pretty straightforward.

olofk, 2019-09-04:
Have you seen the support for loading arbitrary tcl files by setting the file type to `tclSource` ? There's an example of how this is used for the ISE backend to set a TCL command that turns on a different HDL parser [here](https://github.com/fusesoc/blinky/blob/master/blinky.core#L50)

Would that help in your case?

GCHQDeveloper560, 2019-09-04:
I believe I've seen that before and promptly forgot so thanks for the reminder! I think that will work for this case.

Unfortunately I'd already gotten ahead of myself! See GCHQDeveloper560/edalize@43a36203f12e62dc57dd5c4c5a82beef33bbff60. The second commit adding this feature is probably not useful. However, the first commit (0728f4dd79d5b04bd6b9b8f5fa08f2641a1665fe) just converts the existing behaviour to a template. I put it out there in case you think it's a little cleaner than the TCL approach, but it probably just exchanges messy TCL for messy Jinja. I'm fine with taking the change or leaving it!

olofk, 2019-10-14:
Have you had the chance to check if using `tclSource` helps with your case?

Regarding the TCL to jinja conversion I'm still undecided. I have not had a purely positive experience of converting to jinja so I'm kind of leaning towards keeping the TCL as is. But now I know where to find a patch if I was to change my mind

GCHQDeveloper560, 2019-10-16:
Using `tclSource` works for me, and I'm fine with leaving the TCL out of Jinja if that's your preference.

closed 2019-11-21 by olofk