olofk/edalize #50: Vivado - reading data files.

m-kru, 2019-09-13:
It looks like right now it is not possible to read data files into Vivado projects. I think new entry is needed in `file_types` table. Check documentation:
[image]
Please note. that there are more than one possible file extensions.

m-kru, 2019-12-05:
@olofk ping.
The more I use fusesoc, the more it annoys me. I feel like memory configuration files are part of any non academic project. I do not want to write tcl wrappers for this common all the time :(.

olofk, 2019-12-05:
Sorry. Completely missed this one. I don't think I have used read_mem myself so I'm not sure how it works. Are these files standard verilog hex files that can be read with $readmemh? In that case you should just be able to set file_type to user and use $readmemh. Or are these some Xilinx-specific files? In that case I agree that the only sensible thing would be to add a new file type. Not sure what to call it. If other FPGA tools have similar things, which I think they do, then it would make sense to call the file type something generic, like `memoryFile` and let the different backends do what they need for the files they support.

olofk, 2019-12-10:
I just accepted https://github.com/olofk/edalize/pull/86 which hopefully fixes this by adding file type `mem`. Please close or comment when you have had a chance to try it out