olofk/edalize #51: FuseSoC does not exit after ModelSim simulation ends

a-gibson, 2019-09-18:
When running FuseSoC with a simple core file that includes one VHDL source file and one VHDL test bench with a `sim` target that uses ModelSim, FuseSoC doesn't exit when the simulation comes to an end. Instead, I am presented with a ModelSim prompt, for example:
```
VSIM 2>
```
The ModelSim prompt is unresponsive. Thus the only way to stop FuseSoC and regain my terminal is to kill it, for example `Ctrl-C`, and FuseSoC will report that it's aborted.
```
INFO: ****************************
INFO: ****   FuseSoC aborted  ****
INFO: ****************************
```

I don't know if there is an intention to have a working ModelSim prompt after a simulation finishes or whether FuseSoC should recognise that the simulation has finished and it should exit accordingly. From my perspective I expected the latter, but I am trying to integrate FuseSoC into an automated build flow so a ModelSim prompt is not helpful.

I have found a reasonable workaround by adding the following to my FuseSoC core file, but suspect it isn't a good workaround because it has implications when starting the ModelSim GUI:
```
vsim_options: [-do, exit]
```

Looking at the FuseSoC output I see the following:
```
vsim -c -do "do edalize_main.tcl; exit"
```
I see that there is a clear `exit` included here.

But later on when invoking `vsim` for my test bench the following is used:
```
vsim -do "run -all" -c <library>.<test-bench-entity>
```

Should the last `vsim` invocation also include an `exit`?

olofk, 2019-09-18:
It certainly should exit. Strangely enough, I just tried to change the last invocation to `"run -all; exit"`on my machine and it still doesn't exit. Could you check if that helps in your case? Easiest is probably to just edit the generated Makefile and run `make run`

a-gibson, 2019-09-18:
As you suggested, I changed the following like in the `Makefile`:
```
$(VSIM) -do "run -all" -c $(addprefix -pli ,$(VPI_MODULES)) $(TOPLEVEL) $(EXTRA_OPTIONS)
```
to:
```
$(VSIM) -do "run -all; exit" -c $(addprefix -pli ,$(VPI_MODULES)) $(TOPLEVEL) $(EXTRA_OPTIONS)
```
and when invoked with `make run` ModelSim does exit (with the usual summary of end time, elapsed time and number of warnings and errors) and FuseSoC also exits.

It's odd that our experiments yield different behaviour.

olofk, 2019-09-19:
I think there is something odd with my installation, so if it works for you then let's go with that. I pushed a fix now. Please test and close this or let me know if it's still not working

a-gibson, 2019-09-19:
That fix (f656a6897740359605a5270566359bd2739fe32e) works for me. :-)
Thanks!

closed 2019-09-19 by a-gibson