olofk/edalize #56: Support message / warning reporting

mithro, 2019-10-10:
The various tools can produce a lot of messages about the design. Extract them and report in a useful common format.

mithro, 2019-10-10:
Useful references;
 * https://github.com/leaflabs/xmsgs-tools
 * https://github.com/Nic30/hwtBuildsystem/tree/master/hwtBuildsystem/vivado
 * https://git.elphel.com/Elphel/vdt-plugin/tree/master/parsers

wallento, 2020-06-18:
Hi,
I am not sure this directly relates, but for the new Fusesoc GitHub Actions I am working on (for example see: https://github.com/lowRISC/ibex/pull/960) I have started this one now: https://github.com/librecores/eda-log-parser
I have discussed with Wilson that we want to have a Verilator option to emit messages in a structured way (json?) and @pgielda just recently told me that the same will happen for Verible (hopefully the same format). Anyhow, we still need it for all the other tools (especially proprietary).
The code is a mockup at the moment, but its maybe a starting point and relates to this discussion.
Cheers,
Stefan

pgielda, 2020-06-18:
@wallento is more a discussion now on how to handle this across different tools. But of course it would be great to have a common approach as I mentioned.