olofk/edalize #67: Vivado pgm fails ugly if no device is connected

imphil, 2019-11-01:
If no device is connected, we get this error message. We should tidy it up to be more helpful and less of a stacktrace.

```
vivado -quiet -nolog -notrace -mode batch -source fusesoc_utils_blinky_0_pgm.tcl -tclargs xc7a100tcsg324-1 fusesoc_utils_blinky_0.bit
FuseSoC Xilinx FPGA Programming Tool
====================================

INFO: Programming part xc7a100tcsg324-1 with bitstream fusesoc_utils_blinky_0.bit
INFO: [Labtools 27-2285] Connecting to hw_server url TCP:localhost:3121
INFO: [Labtools 27-2222] Launching hw_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx hw_server v2018.2
  **** Build date : Jun 14 2018-20:18:37
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

ERROR: [Labtoolstcl 44-199] No matching targets found on connected servers: localhost
Resolution: If needed connect the desired target to a server and use command refresh_hw_server. Then rerun the get_hw_targets command.
ERROR: [Common 17-39] 'get_hw_targets' failed due to earlier errors.

    while executing
"get_hw_targets"
    invoked from within
"foreach { hw_target } [get_hw_targets] {
    puts "INFO: Trying to use hardware target $hw_target"

    current_hw_target $hw_target

    # Open hardw..."
    (file "fusesoc_utils_blinky_0_pgm.tcl" line 17)
make: *** [Makefile:22: pgm] Error 1
ERROR: Failed to run fusesoc:utils:blinky:0 : 'make' exited with an error code
```

I thought `-notrace` should have taken care of the `while executing...` part. But perhaps this is because it's a nested error? Pulling this anyway and perhaps there are some more improvements to be done later on

_Originally posted by @olofk in https://github.com/olofk/edalize/pull/65#issuecomment-548773135_

olofk, 2019-12-22:
I wonder if this might work

```diff
--- a/edalize/templates/vivado/vivado-program.tcl.j2
+++ b/edalize/templates/vivado/vivado-program.tcl.j2
@@ -14,7 +14,11 @@ connect_hw_server
 
 # Find the first target and device that contains a FPGA $part.
 set hw_device_found 0
-foreach { hw_target } [get_hw_targets] {
+if {[catch get_hw_targets hw_targets]} {
+    exit 1
+}
+
+foreach hw_target $hw_targets {
     puts "INFO: Trying to use hardware target $hw_target"
 
     current_hw_target $hw_target

```

Don't have access to a board right now to test the working case but it seems to fail nicer at least :)

imphil, 2019-12-23:
I can give it a try in the new year.

olofk, 2020-03-16:
Are we happy with it now?

andmeyee, 2020-12-07:
Is there any progress on this issue? The generation of a bitstream file without the need to program a connected device would be nice.

olofk, 2020-12-07:
I agree. In hindsight it was a questionable choice to try and program boards by default and most of the new FPGA backends don't have this functionality. In the long run I think there should be some extra config in the target sections to define how and if the board should be programmed. There's also some more thoughts about this [here](https://github.com/olofk/fusesoc/issues/319).

Oh, wait a minute. The easiest thing you can do right now is to add `--build` to the command parameters. That way FuseSoC will stop after creating the bitstream

andmeyee, 2020-12-17:
Ok, that helps for the moment. Thanks a lot for the hint.