olofk/edalize #84: VUnit support: vhdl_standard should be of type 'str'

krishnan-gopal, 2019-12-09:
The latest commit of VUnit, as well as the last release v4.3.0 both expect that the vhdl_standard for a source-file in the run.py config file should be of type 'str' and not 'VHDL.standard'.

Currently, the generated run.py uses the type 'VHDL.standard', and I am getting the error:

Traceback (most recent call last):
  File "*****/run.py", line 25, in <module>
    lib.add_source_files("*****/sourcefile.vhd", vhdl_standard=VHDL.standard("2008"))
  File "*****/vunit/vunit/ui/library.py", line 216, in add_source_files
    for file_name in file_names
  File "*****/vunit/vunit/ui/library.py", line 216, in <listcomp>
    for file_name in file_names
  File "*****/vunit/vunit/ui/library.py", line 270, in add_source_file
    vhdl_standard=self._which_vhdl_standard(vhdl_standard),
  File "*****/vunit/vunit/ui/library.py", line 368, in _which_vhdl_standard
    return VHDL.standard(vhdl_standard)
  File "*****/vunit/vunit/vhdl_standard.py", line 80, in standard
    return VHDLStandard(name)
  File "*****/vunit/vunit/vhdl_standard.py", line 27, in __init__
    if standard.endswith(standard_name) and len(standard_name) == 2:
TypeError: endswith first arg must be str or a tuple of str, not VHDLStandard

Does this need to be corrected or I am missing something here ?

cmarqu, 2019-12-14:
Pinging @julian-becker who added the VUnit support.

julian-becker, 2019-12-15:
When I did the original implementation, I was developing against vunit 4.2.0. Since then, it seems there have been some subtle changes to this API. The changes in #88 should fix this, but unfortunately break compatibility for vunit 4.2.0...

olofk, 2019-12-20:
@kraigher @larsasplund Do you have any comments on this? Did the API change in a backwards-incompatible way or is there a way to support both older and newer VUnit?

kraigher, 2019-12-20:
The Public API has always been str. Internally we switched to an object to get a better implementation of some rules based on the standard. In 4.2.0 there was a bug where the Public API accepted the VHDLStandard object which it should not have. In 4.3.0 we fixed this http://vunit.github.io/release_notes.html

So in summay VHDLStandard is a private class that should thus never appear on the public api

olofk, 2019-12-21:
Cool. Thanks @kraigher for clearing that up. @krishnan-gopal, @julian-becker is there anything we can do in the Edalize backend to use the public API instead? I haven't had time to look at the implementation myself unfortunately

olofk, 2019-12-21:
Or is https://github.com/olofk/edalize/pull/88 the correct fix that will work on all versions except for 4.2.0?

kraigher, 2019-12-21:
Well 4.2.0 could be considered broken so maybe you do not need to support it. Also I would recommend always using the public api since that is the only thing we care about being stable.

julian-becker, 2019-12-21:
Thanks @kraigher, for the additional support on this issue!
My observation is that on both vunit_hdl 4.2.0 and 4.1.0, I get the following error when using the `vhdl_standard` argument as string:

    Compiling into lib:       ..\src\vunit_examples-vhdl_run_0\test_control.vhd                                                   Traceback (most 
    recent call last):
    File "c:\progs\python37\lib\site-packages\vunit\ui.py", line 838, in main
        all_ok = self._main(post_run)
    File "c:\progs\python37\lib\site-packages\vunit\ui.py", line 880, in _main
        return self._main_compile_only()
    File "c:\progs\python37\lib\site-packages\vunit\ui.py", line 1012, in _main_compile_only
        self._compile(simulator_if)
    File "c:\progs\python37\lib\site-packages\vunit\ui.py", line 1044, in _compile
        printer=self._printer)
    File "c:\progs\python37\lib\site-packages\vunit\simulator_interface.py", line 172, in compile_project
        self.compile_source_files(project, printer, continue_on_error)
    File "c:\progs\python37\lib\site-packages\vunit\simulator_interface.py", line 244, in compile_source_files
        if self.__compile_source_file(source_file, printer):
    File "c:\progs\python37\lib\site-packages\vunit\simulator_interface.py", line 189, in __compile_source_file
        command = self.compile_source_file_command(source_file)
    File "c:\progs\python37\lib\site-packages\vunit\modelsim_interface.py", line 138, in compile_source_file_command
        return self.compile_vhdl_file_command(source_file)
    File "c:\progs\python37\lib\site-packages\vunit\modelsim_interface.py", line 162, in compile_vhdl_file_command
        + [self._std_str(source_file.get_vhdl_standard()), '-work', source_file.library.name, source_file.name])
    File "c:\progs\python37\lib\site-packages\vunit\modelsim_interface.py", line 151, in _std_str
        if vhdl_standard <= VHDL.STD_2008:
    File "c:\progs\python37\lib\functools.py", line 105, in _ge_from_lt
        op_result = self.__lt__(other)
    File "c:\progs\python37\lib\site-packages\vunit\vhdl_standard.py", line 39, in __lt__
        return int(self._standard) < int(other._standard)  # pylint: disable=protected-access
    AttributeError: 'str' object has no attribute '_standard'
    ERROR: Failed to build vunit:examples-vhdl:run:0 : 'c:\progs\python37\python.exe' exited with an error code

With 4.3.0, 4.0.0 and 3.9.0, the string version works just fine, and I think we should probably stick with it as it is the proper API usage and since - as you said - 4.2 could be considered broken in this respect.

olofk, 2019-12-22:
I decided to pull in https://github.com/olofk/edalize/pull/88 by @julian-becker which hopefully should fix this issue. @krishnan-gopal can you confirm this works for you and close the issue if it does?

krishnan-gopal, 2020-01-02:
Ok, i just tested this now and it works. Thanks a lot.

closed 2020-01-02 by krishnan-gopal