olofk/edalize #89: Support Synopsys Design Compiler (DC) for Synthesis

imphil, 2019-12-20:
We need support for Synopsys Design Compiler for synthesis. This will most likely be a rather minimal backend, where only the filelist is coming from edalize, all other configuration is done by specifying a TCL file with the appropriate flow.

msfschaffner, 2019-12-20:
@olofk, if you need someone to help with this (including running some tests) I can assist.

olofk, 2019-12-21:
That would be excellent. DC is one of the most requested tools. I very much appreciate the offer to help since I don't have access to the tool myself

msfschaffner, 2020-01-14:
Ok so DC is a bit tricky, since in addition to the tool flow itself, you need to set up technology specific libraries to make this work. Hence, it is in general not possible to provide a "default" flow that will just run out of the box. The user always has to supply a minimum amount of info about his technology setup.

The flow could be built on top of what is provided for AscentLint at the moment. 

It would be great to support three modes here: 
1. batch
2. interactive shell
3. gui

For 1. and 2., the tool can be invoked as follows (just the exit statement should not be inserted into the runfile for 2.)
```bash
TERM="dtterm" dc_shell-xg-t $@ -f synth.tcl | tee synth.log
```
For 3., and additional `-gui` switch is required
```bash
TERM="dtterm" dc_shell-xg-t $@ -f synth.tcl -gui | tee synth.log
```
The terminal setting is a bit peculiar, but required such that the DC shell works correctly.

The main run file is `synth.tcl` and is meant to contain a simple synthesis flow.
```tcl
#####################
##  PREPARE FLOW   ##
#####################

# this is a setup script that the user needs to provide (see template below). 
source setup.tcl

# make library
sh mkdir -p $WORKLIB
define_design_lib WORK -path $WORKLIB

#######################
##  DESIGN SOURCES  ###
#######################

# define the top level 
set TOP_ENTITY <TOP_ENTITY_NAME>

# read source list generated by fusesoc
set fp [open [ls "*.scr"] r]
set file_data [read $fp]
close $fp
set data [split $file_data "\n"]

# parse out the SV defines that Fusesoc puts into the source list
set SRC {}
set DEFINE ""
foreach item $data {
	if {[string range $item 0 7] == "+define+"}  {
		set DEFINE "${DEFINE}[string range $item 8 end] "
	} else {
		lappend SRC $item
	}
}

# additional parameter override (empty at the moment, 
# could be passed through by fusesoc as well in a similar way as +define)
set PARAMS ""

###########################
##   ELABORATE DESIGN    ##
###########################

# delete previous designs.
remove_design -designs
sh rm -rf $WORKLIB/*

analyze -define ${DEFINE} -format sv ${SRC}
elaborate  ${TOP_ENTITY} -parameters ${PARAMS}
[... 113 lines cut ...]

# timing constraint in ns 
set TCK     10.0
set IN_DEL  2.0
set OUT_DEL 2.0
set DELAY   ${TCK}

# define a primary clock
create_clock ${CLK_PIN} -period ${TCK}

# assume these nets are ideal during synthesis
set_ideal_network ${CLK_PIN}
set_ideal_network ${RST_PIN}

# constrain all combinatorial paths
set_max_delay ${DELAY} -from [all_inputs] -to [all_outputs]
set_input_delay ${IN_DEL} [remove_from_collection [all_inputs] {${CLK_PIN}}] -clock ${CLK_PIN}
set_output_delay ${OUT_DEL}  [all_outputs] -clock ${CLK_PIN}

# input/output pins of stdcells are typically labelled with A,X,Y,Z or similar
set_driving_cell  -no_design_rule -lib_cell ${driving_cell} -pin X [all_inputs]
set_load [load_of ${load_lib}/${load_cell}/A] [all_outputs]
```

I still need to think about how we can check for success at the end of the flow.

msfschaffner, 2020-01-14:
Let me know if you need more info or whether I should test something.

I still need to think about how we can check for success at the end of the flow...