olofk/edalize #91: Allow DSE to be used for Quartus P&R

DHWinterWolf, 2019-12-23:
I use DSE quite a bit to handle multiple P&R runs in parallel for some complex designs targeting Altera FPGAs.

It looked like it should be reasonably straightforward to use the pnr tool option for Quartus to allow DSE to be used instead of the standard Quartus P&R flow, so I had a go at implementing it for my requirements (a seed sweep with limits on parallel runs). It worked a treat, taking options from a FuseSoC core and producing exactly the result I was expecting.

I'm hoping to tidy this up over the next few days, make it completely configurable for other DSE flow types and add a few tests, then submit a PR in case it is of interest to anyone else. Any comments or suggestions are welcome, of course!

DHWinterWolf, 2020-01-03:
I've now added tests for this to the PR as well - feel free to shout if anything else is missing to bring this up to standards.

a-gibson, 2020-01-08:
I've used DSE in the past to speed up CI runs which require a few attempts to find a solution that achieves timing closure.

If FuseSoC could support DSE this will be very helpful in future!

olofk, 2020-01-23:
Of course FuseSoC supports DSE :)

closed 2020-01-23 by olofk