olofk/edalize #92: Names of generated .xpr, .bit etc. files.

m-kru, 2019-12-29:
Based on Fusesoc + Vivado experience, but it is probably the same for other back ends.
Names of generated project files follows this pattern: {CORE_NAME}_{VERSION}.extension. When working with multiple targets for the same core it is not always clear with which target you work. For example, you open Vivado and based on the project name you do not know what target it is. The same story when you want to program device and you choose .bit file. I think it might be helpful if these names include also target name, if it is different than `default`.  These names would be a combination of core name, version and target name. @olofk what do you think?

olofk, 2020-01-01:
I see your point, and we could perhaps change this, but there is also another way this can be solved. You can specify `--system-name` to set a custom name for these files. The work root of the project also contains the target name, so it should be possible to look at the path of the project files to figure out which target you are currently looking at.

If this isn't enough we could look at changing this, but I know that people might use hard coded paths in external scripts and this will break things for them so I hope to avoid changing if we don't have to

m-kru, 2020-01-03:
One more idea came to my mind, that does not break backward compatibility.
An option to specify `system-name` key in the `.core` file. If given target has `system-name` key then use value of this key as a name. If it doesn't use the default one (the one that is used right now). One inconsistency that I can see is that in `.core` files you use floor `_` instead of hyphen `-`, so maybe it would be `system_name`. What do you think @olofk ?

olofk, 2020-01-23:
That's a good idea. We need to look inside FuseSoC to see how hard it would be to implement. Hopefully very easy but there could be something I have missed.