olofk/edalize #106: Force forward slash paths - Verilator on Cygwin

AlistairSymonds, 2020-02-17:
A bit a niche circumstance I admit, but I sometimes do some work ona windows desktop via a Cygwin environment with verilator and GTKWave installed as described here: https://zipcpu.com/blog/2017/07/28/cygwin-fpga.html

Now when building with Verilator it gives this error - at first glance its pretty obvious that the paths are messed up, namely the backslashes are not valid path seperators for the makefile. 

```
$ fusesoc --cores-root . run --target=sim pawc
INFO: Preparing ::elf-loader:1.0.2
INFO: Preparing ::hyperram:0
INFO: Preparing ::verilator_tb_utils:0
INFO: Preparing ::picorv32:0
INFO: Preparing ::pawc:0.1

verilator -f pawc_0.1.vc
%Error: Cannot find file containing module: ..srchyperram_0hyper_xface.v
        ... Looked in:
             ..srcelf-loader_1.0.2/..srchyperram_0hyper_xface.v
             ..srcelf-loader_1.0.2/..srchyperram_0hyper_xface.v.v
             ..srcelf-loader_1.0.2/..srchyperram_0hyper_xface.v.sv
             ..srcverilator_tb_utils_0/..srchyperram_0hyper_xface.v
             ..srcverilator_tb_utils_0/..srchyperram_0hyper_xface.v.v
             ..srcverilator_tb_utils_0/..srchyperram_0hyper_xface.v.sv
             ..srchyperram_0hyper_xface.v
             ..srchyperram_0hyper_xface.v.v
             ..srchyperram_0hyper_xface.v.sv
%Error: Cannot find file containing module: ..srchyperram_0hyper_dword.v
%Error: Cannot find file containing module: ..srcpicorv32_0axi4_memory.v
%Error: Cannot find file containing module: ..srcpicorv32_0picorv32.v
%Error: Cannot find file containing module: ..srcpicorv32_0picorv32_top.v
%Error: Cannot find file containing module: ..srcpawc_0.1rtl/c10lp_baseline_pinout.v
%Error: Cannot find file containing module: ..srcpawc_0.1rtl/pawc_top.sv
%Error: Cannot find file containing module: ..srcpawc_0.1tb/pawc_tb.sv
%Error: Exiting due to 8 error(s)
make: *** [Makefile:16: Vpawc_tb.mk] Error 1
ERROR: Failed to build ::pawc:0.1 : 'make' exited with an error code

```

Here is the generated .vc file looks like:

```
--Mdir .
--cc
-LDFLAGS -lelf
+incdir+..\src\elf-loader_1.0.2
-CFLAGS -I..\src\elf-loader_1.0.2
+incdir+..\src\verilator_tb_utils_0
-CFLAGS -I..\src\verilator_tb_utils_0
..\src\hyperram_0\hyper_xface.v
..\src\hyperram_0\hyper_dword.v
..\src\picorv32_0\axi4_memory.v
..\src\picorv32_0\picorv32.v
..\src\picorv32_0\picorv32_top.v
..\src\pawc_0.1\rtl/c10lp_baseline_pinout.v
..\src\pawc_0.1\rtl/pawc_top.sv
..\src\pawc_0.1\tb/pawc_tb.sv
--top-module pawc_tb
--exe
..\src\elf-loader_1.0.2\elf-loader.c
..\src\verilator_tb_utils_0\verilator_tb_utils.cpp
..\src\verilator_tb_utils_0\jtagServer.cpp
..\src\picorv32_0\picorv32_tb.cpp
```

And to fix it a find+replace of '\\' with '/' then calling make seems to do the job, but obviously a manual step like that is a bit of a hack. I'll be honest this might more be a leakage on the part of Cygwin (purposefully) not being a perfect abstraction so feel free to close if its not EDAlize's area :P

olofk, 2020-02-19:
Let's get that fixed. Over time I have come to realize that most of the EDA tools actually only use forward slashes even on Windows. I'd say it's a blessing in the greater scheme of things but it causes some edge cases. Would highly appreciate if someone could step up and tell me how to do a proper cygwin detection and what the situation is on e.g. WSL

AlistairSymonds, 2020-02-20:
WSL works fine - its far more isolated from the rest of the windows system, eg when you access files on the Windows part its via a mount of your whole drive in Linux as opposed to Cygwin which (I think) just accesses it directly through the windows system calls dressed up like Linux ones.

Would wlecome any input from someone more familar with the OS internals than I however :)

olofk, 2020-02-21:
This got me wondering if there are any cases at all where verilator needs back slash. Perhaps we can just force forward slashes and not to any platform detection at all. Any clues @wsnyder ?

wsnyder, 2020-02-23:
Verilator always assumes forward slashes separate paths.  A backslash in a file list (.vc) file is typically used to quote a quote, e.g. "\""), which Verilator understands, as do several but not all commercial simulators.

olofk, 2020-03-04:
Thanks @wsnyder! @AlistairSymonds, I just pushed a patch for edalize now. Please let me know if that helps

olofk, 2020-11-17:
I believe this one can be closed. Reopen if problem remains

closed 2020-11-17 by olofk