olofk/edalize #107: ModelSim exit codes on failure are always 0

GCHQDeveloper992, 2020-02-20:
As part of integrating FuseSoC & Edalize with our continuous integration setup, I've noticed that ModelSim always has an exit code of 0 when tests fail with either:

- A REPORT with severity ERROR
- A call to std.env.finish, passing a non zero error count
- A call to std.env.stop, passing a non zero error count

It appears that ModelSim treats the completion of a simulation, regardless of errors/warnings, as a success. 

In a previous CI setup, I remember using the 'onbreak' macro to set a flag that could be passed to the 'quit' functions '-code' argument which allowed CI to detect a non zero exit code successfully. I've made an example patch [on my fork](https://github.com/GCHQDeveloper992/edalize/tree/patch/return-error-codes-from-modelsim) which implements something similar. 

Do you think this is a reasonable way to support returning error codes for easier CI integration? I'll raise a pull request if this seems sensible.

a-gibson, 2020-02-21:
I've been integrating FuseSoC/ModelSim flows into my CI system and hadn't noticed this, which is kinda worrying as I may not be detecting errors! Will need to go have a look.

I presume a `REPORT` with `SEVERITY FAILURE` also causes ModelSim to exit with a code of 0? I know you can change some attributes around this in the `modelsim.ini` file.

It's seems a shame to have to write a wrapper around this but if ModelSim doesn't provide a convenient way to exist with a non-zero status then I think your option is as good as any.

a-gibson, 2020-02-21:
As I suspected, increasing the severity made no difference.

I have stumbled across this:
```
quit -code [coverage attribute -name TESTSTATUS -concise]
```

So the existing line in the `Makefile` could be modified from:
```
$(VSIM) -do "run -all; exit" -c $(addprefix -pli ,$(VPI_MODULES)) $(EXTRA_OPTIONS) $(TOPLEVEL)
```

to:
```
$(VSIM) -do "run -all; quit -code [coverage attribute -name TESTSTATUS -concise]" -c $(addprefix -pli ,$(VPI_MODULES)) $(EXTRA_OPTIONS) $(TOPLEVEL)
```

Would that simplify things? I've only done a very quick test by hacking a Makefile so this maybe doesn't account for all scenarios.

olofk, 2020-02-21:
Agree. This is awkward. I scanned the vsim options for any relevant settings but came up blank as well.

If there's a tcl command to do this, an alternative would be to add this to a tcl file and load it in the core file with `tclSource`. The benefit would be that it allows flexibility to break on failure, error or even warning to the discretion of the testbench designer. The drawback would be that it needs to be included in all designs that need it

GCHQDeveloper992, 2020-03-02:
Thanks to both of you for the insight. I quite like the approach that @a-gibson has suggested, although whilst trying to it doesn't seem to cover cases where the simulator has quit because of std.env calls. However, for what I'm trying to achieve doing something like that combined with a suitable REPORT and SEVERITY call before the quit would work well.

One thing I might suggest is adding a small condition on the quit code - I don't think we want to cause an error when someone has used SEVERITY WARNING in their code (I can think of instances where that is used that I wouldn't want to cause an error). Perhaps we could change the execution to:

`$(VSIM) -do "run -all; quit -code [expr [coverage attribute -name TESTSTATUS -concise] >= 2 ? [coverage attribute -name TESTSTATUS -concise] : 0]" -c $(addprefix -pli ,$(VPI_MODULES)) $(EXTRA_OPTIONS) $(TOPLEVEL)`

The double call to coverage looks odd, but I just can't convince the `set` command to behave. I might push that up as a PR request? @olofk - the tclSource idea is interesting, but there could be a huge number of designs to add that to?

a-gibson, 2020-03-13:
Are there any objections to merging this?

I would also find this beneficial for the same reasons as mentioned in #121 - it's really helpful for CI tools like Jenkins to know when a test has failed so it can take the right actions and report corresponding results.

olofk, 2020-03-16:
I merged this today. Had to add a follow-up patch though to update the reference files for the test suite. Please remember to run pytest and make sure the tests pass

closed 2020-03-16 by olofk