olofk/edalize #119: AscentLint tool switches

msfschaffner, 2020-03-11:
We are using the Fusesoc AscentLint target, and run into issues with our license queue when running batch regressions. In particular, without passing the `-wait_license` switch to the tool, it immediately errors out with a FlexNet license error, which makes runs unstable.

Would it be possible to either add a simple binary flag for this, or even better, a runtime parameter string that could optionally passed to the tool here:
https://github.com/olofk/edalize/blob/36c4b0c01d02fc7571dc3099187572a9ba2db81d/edalize/templates/ascentlint/Makefile.j2#L8

CC @imphil @sriyer

olofk, 2020-05-04:
Sorry. Completely missed this one. Yes, being able to pass more command line options to ascentlint would be my preferred solution. Most tools support that already.

Would something like [this](https://github.com/olofk/edalize/tree/ascentlint_options) work? Running `fusesoc run --tool=ascentlint serv --ascentlint_options=-wait_license` seems to do the right thing, but I don't have ascentlint here to verify

imphil, 2020-05-04:
Oh, missed this one as well. Yes, your patch looks correct @olofk and in line with what the other backends do.

msfschaffner, 2020-05-04:
Thanks @olofk. I gave this a quick spin and it looks correct.

olofk, 2020-06-17:
@msfschaffner @imphil  Should I apply the patch as is?

imphil, 2020-06-17:
Yes, please do.

To make sure I have less chance of missing that, please open a PR and ask a review from me, I try to go through those lists regularly.

msfschaffner, 2020-06-19:
Yes it seems to be working as-is.

msfschaffner, 2020-07-13:
I've hijacked @olofk's commit and PR'd it. Let me know if something else needs to change.

commit ab7fb59f (2020-07-14): Add support for ascentlint_options

Support tool options (runtime parameters) in AscentLint.

Fixes  #119
closed 2020-07-14 by imphil