olofk/edalize #126: Parsing backend arguments truncates strings

DHWinterWolf, 2020-03-21:
I'm a bit unsure whether the problem here lies with FuseSoC or Edalize, but I think it's probably Edalize. I see that both products have had some changes in the last month or so to do with parsing arguments. I found the bug when using FuseSoC. If you do something like this:

```
fusesoc run --target <something> v:l:n:v
```

Then everything seems fine. The core file is parsed, and arguments in the core file are used to configure Edalize to run in the way you'd expect. In my particular case, by target ran Quartus and had "pnr: none" in the core file to skip place and route. This also skips pretty much all of the run step, as there's nothing to program. However, if you do this (as a testing system we have does):

```
fusesoc run --target <something> v:l:n:v --pnr none
```

For the same target and the same core, the run section tries to program a file.

I've done some digging around, and it looks like this comes down to FuseSoC parsing arguments itself for some things (including the build stage, so "--pnr none" does seem to avoid placing and routing the design), but then passes the command line arguments object to Edalize to re-parse itself for run. This is why I'm not sure if it's FuseSoC or Edalize with the issue - should it be doing that?

I'm assuming it should, in which case I've looked at the parsing. It seems after some runs trying to narrow things down that that Edalize uses a different method for parsing the arguments, which causes strings to be truncated to just their first character. As such, it ended up with `pnr` set to `n`, rather than `none`.

I changed the argument parsing to be more like FuseSoC - the changes are available on a fork at https://github.com/olofk/edalize/commit/695cc30854fddedcc7636c8f627148685e420dfd

With this change in Edalize, passing `--pnr <mode>` works as expected. The reason for not raising a PR is that I wanted to check with others whether this is the right place to be doing this. What do people think?

olofk, 2020-03-21:
Ok, this sounds bad. I haven't had time to look at it yet though.

In general, the end goal is to move all argument parsing out from edalize (and into FuseSoC). Thought I had a good plan https://github.com/olofk/fusesoc/issues/363 but it's a bit fiddly so I'm not surprised if something went wrong.

DHWinterWolf, 2020-03-22:
> Ok, this sounds bad. I haven't had time to look at it yet though.
> 
> In general, the end goal is to move all argument parsing out from edalize (and into FuseSoC). Thought I had a good plan [olofk/fusesoc#363](https://github.com/olofk/fusesoc/issues/363) but it's a bit fiddly so I'm not surprised if something went wrong.

Right, this makes the intention a lot clearer in my mind now, thanks for pointing me at that. In that case, I think the solution is a lot more simple, and becomes a change in FuseSoC. I'll get a diff pushed, mention it on your issue on FuseSoC and close this issue.

DHWinterWolf, 2020-03-22:
Closing in response to opening a PR against FuseSoC: https://github.com/olofk/fusesoc/pull/380

closed 2020-03-22 by DHWinterWolf