olofk/edalize #144: Go Python 3.5+-only

imphil, 2020-04-27:
Python 2.7 has reached EOL earlier this year, and fusesoc has removed support for it and is Python 3.5-only now. The relevant discussion is in https://github.com/olofk/fusesoc/issues/337, most arguments apply to edalize as well.

As noted before: nothing prevents users on Python 2.7 to continue using edalize 0.2. This change only affects people going forward, but does not break existing setups.

`pypistats python_minor edalize`

```
| category | percent | downloads |
|----------|--------:|----------:|
| 3.7      |  74.87% |    24,014 |
| 3.5      |  12.37% |     3,966 |
| 3.6      |   9.08% |     2,912 |
| 2.7      |   1.54% |       494 |
| null     |   1.14% |       367 |
| 3.8      |   0.98% |       314 |
| 3.4      |   0.02% |         7 |
| Total    |         |    32,074 |

Date range: 2019-10-29 - 2020-04-26
```

olofk, 2020-04-27:
Yes, let's drop it

GCHQDeveloper560, 2020-04-28:
I've added this to #143 and it now has the magical green check mark.

commit 6f3cfeb8 (2020-04-30): Remove Python 2 support

Fixes #144
closed 2020-04-30 by imphil