olofk/edalize #158: Add `include_path` property to include files

tudortimi, 2020-05-16:
Needed to support `include` directives that contain paths, not just simple file names (as discussed at https://github.com/olofk/fusesoc/issues/397):

```
`include "some_dir/some_file.svh"
```

`some_file.svh` would have the properties `is_include_file` and `include_path: some_dir`.

`include_path` has to be a relative path starting from the core root (where the `.core` file is located), which points to a directory. We could also check that that directory is the parent or a grandparent directory of the file on which the property is attached.

olofk, 2020-05-16:
Agree. Specifying the include path as relative to core_root seems to make modt sense.

I think we can skip checking if it's a parent dir. That way we can probably also support things like `include ../dir/file.vh` in case someone does that.

Question: do we allow setting include_path without `is_include_file : true`. I think we can allow it but ignore include_path in that case, and let the frontend (e.g. FuseSoC) emit a warning

tudortimi, 2020-05-17:
Why not an error if `include_path` is set on a file that is not marked with `is_include_file: true`?

olofk, 2020-05-17:
I don't know really. Can probably come up with some convoluted example but don't have a real use case. Just think we don't have to make it illegal in case we realize later on that a valid use case exists

tudortimi, 2020-05-17:
Isn't it easier (in terms of keeping backward compatibility) to relax existing restrictions than it is to add new ones? Plus: https://en.wikipedia.org/wiki/You_aren%27t_gonna_need_it

olofk, 2020-05-28:
I believe we can close this now

closed 2020-05-28 by olofk