olofk/edalize #163: Tests failing

FFY00, 2020-05-30:
Hey, I am packaging this for archlinux and the tests are failing. I am building from the latest release.

Are the tools needed for the tests? Some of them seem to pass without them installed.

The test output is also a bit fiddly, sometimes `test_icarus.py` fails, sometimes it doesn't.

```
============================= test session starts ==============================
platform linux -- Python 3.8.3, pytest-5.4.2, py-1.8.1, pluggy-0.13.1
rootdir: /build/python-edalize/src/edalize-0.2.2
collected 41 items

tests/test_ascentlint.py .                                               [  2%]
tests/test_diamond.py ..                                                 [  7%]
tests/test_edam.py ......                                                [ 21%]
tests/test_ghdl.py .                                                     [ 24%]
tests/test_icarus.py F.                                                  [ 29%]
tests/test_icestorm.py ......                                            [ 43%]
tests/test_ise.py ..                                                     [ 48%]
tests/test_isim.py .                                                     [ 51%]
tests/test_modelsim.py .                                                 [ 53%]
tests/test_quartus.py .                                                  [ 56%]
tests/test_rivierapro.py .                                               [ 58%]
tests/test_spyglass.py FF                                                [ 63%]
tests/test_vcs.py ...                                                    [ 70%]
tests/test_veribleformat.py .                                            [ 73%]
tests/test_veriblelint.py .                                              [ 75%]
tests/test_verilator.py ...                                              [ 82%]
tests/test_vivado.py ...                                                 [ 90%]
tests/test_vunit.py ..                                                   [ 95%]
tests/test_xcelium.py .                                                  [ 97%]
tests/test_xsim.py .                                                     [100%]

=================================== FAILURES ===================================
_________________________________ test_icarus __________________________________

    def test_icarus():
        import os
        import shutil
        from edalize_common import compare_files, setup_backend, tests_dir

        ref_dir      = os.path.join(tests_dir, __name__)
        paramtypes   = ['plusarg', 'vlogdefine', 'vlogparam']
        name         = 'test_icarus_0'
        tool         = 'icarus'
        tool_options = {
            'iverilog_options' : ['some', 'iverilog_options'],
            'timescale'        : '1ns/1ns',
        }

        (backend, work_root) = setup_backend(paramtypes, name, tool, tool_options, use_vpi=True)
        backend.configure()

        compare_files(ref_dir, work_root, ['Makefile',
                                           name+'.scr',
                                           'timescale.v',
        ])

        backend.build()
        compare_files(ref_dir, work_root, ['iverilog.cmd'])
>       compare_files(ref_dir, work_root, ['iverilog-vpi.cmd'])

tests/test_icarus.py:27:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

ref_dir = '/build/python-edalize/src/edalize-0.2.2/tests/test_icarus'
work_root = '/tmp/icarus_tzu23ah2', files = ['iverilog-vpi.cmd']

    def compare_files(ref_dir, work_root, files):
        """Check that all *files* in *work_root* match those in *ref_dir*.
[... 183 lines cut ...]

tests/edalize_common.py:29: AssertionError
----------------------------- Captured stdout call -----------------------------
sg_shell -enable_pass_exit_codes -tcl spyglass-run-design_read.tcl
sg_shell -enable_pass_exit_codes -tcl spyglass-run-lint_lint_rtl.tcl
sg_shell -enable_pass_exit_codes -tcl spyglass-run-some_othergoal.tcl
------------------------------ Captured log call -------------------------------
WARNING  edalize.spyglass:spyglass.py:140 qip_file.qip has unknown file type 'QIP'
WARNING  edalize.spyglass:spyglass.py:140 qsys_file has unknown file type 'QSYS'
WARNING  edalize.spyglass:spyglass.py:140 sdc_file has unknown file type 'SDC'
WARNING  edalize.spyglass:spyglass.py:140 bmm_file has unknown file type 'BMM'
WARNING  edalize.spyglass:spyglass.py:140 pcf_file.pcf has unknown file type 'PCF'
WARNING  edalize.spyglass:spyglass.py:140 ucf_file.ucf has unknown file type 'UCF'
WARNING  edalize.spyglass:spyglass.py:140 xci_file.xci has unknown file type 'xci'
WARNING  edalize.spyglass:spyglass.py:140 xdc_file.xdc has unknown file type 'xdc'
WARNING  edalize.spyglass:spyglass.py:140 bootrom.mem has unknown file type 'mem'
WARNING  edalize.spyglass:spyglass.py:140 c_file.c has unknown file type 'cSource'
WARNING  edalize.spyglass:spyglass.py:140 cpp_file.cpp has unknown file type 'cppSource'
WARNING  edalize.spyglass:spyglass.py:140 config.vbl has unknown file type 'veribleLintRules'
=========================== short test summary info ============================
FAILED tests/test_icarus.py::test_icarus - AssertionError: iverilog-vpi.cmd
FAILED tests/test_spyglass.py::test_spyglass_defaults - AssertionError: spygl...
FAILED tests/test_spyglass.py::test_spyglass_tooloptions - AssertionError: sp...
========================= 3 failed, 38 passed in 1.76s =========================
```

GCHQDeveloper560, 2020-06-05:
The tests don't need the actual tools and instead use mock versions of the tools in `tests/mock_commands`. The GitHub CI setup is just Ubuntu without anything special installed, and tests Python 3.8 so there shouldn't be anything surprising there. Files like `iverilog-vpi.cmd` referenced in the errors are just one-liners with the command line arguments recorded.

A frequent source of test failures like these is command line arguments getting out of order, so they may be valid but don't match the reference output. I don't know why you might be seeing an intermittent failure. Unfortunately the test output doesn't actually show the differences between the two files reference in the error. Are you able to post them?

olofk, 2020-06-17:
Thanks for working on the packaging. Unfortunately I haven't got a clue about what goes wrong here. Ideally the differences between the files should be shown in the output, but thanks to an annoying "feature" in pytest, this doesn't work (happens when the assert is in a different file from the test function).

If you can compare the files between ref_dir and work_root it would help a lot to analzye what goes wrong

FFY00, 2020-06-17:
Sticking a `print(os.listdir(work_root))` shows:

```
['Makefile', 'test_icarus_minimal_0.scr']
```

For `test_icarus_minimal`, there should be a `spyglass.cmd` right?

```
['Makefile', 'spyglass-run-some_othergoal.tcl', 'spyglass-run-lint_lint_rtl.tcl', 'spyglass-run-design_read.tcl', 'test_spyglass_0.prj']
```

For `test_spyglass_tooloptions`. Similarly, there should be a `spyglass.cmd`, right?

GCHQDeveloper560, 2020-06-18:
The test failures print the name of the temporary directory used for the output, so you shouldn't have to dive into Python to check the test's output files. In the example you posted the failing tests put their outputs in `/tmp/icarus_tzu23ah2/iverilog-vpi.cmd`, `/tmp/spyglass_4py1xwbp/spyglass.cmd`, and `/tmp/spyglass_nal23vau/spyglass.cmd`.

For `test_icarus_minimal` I'd expect to see `Makefile`, `test_icarus_minimal_0.scr`, `iverilog.cmd`, and `vvp.cmd`. The two files you show are created from the `configure` command and the later two from `build` and `run`, so I don't know whether your list is from early in the run before they were created or whether something is going wrong in the build phase that is resulting in them not being created. Similarly for the Spyglass test it looks like the `spyglass.cmd` file that should be created by the `build` command is missing.

I may be wrong, but I would expect different errors if the files were totally missing. The earlier failures you posted seem more like the files are there but don't have the expected contents. If that's the case then posting the contents of the failing files from the temporary directory would be helpful.

olofk, 2021-06-02:
Do we know if this is still an issue?