olofk/edalize #167: [Verible] Rename tools

imphil, 2020-06-09:
Verible (thankfully) renamed their binaries to include "verible" in the name. We need to follow in the verible lint and format backends. 

https://github.com/google/verible/commit/a6d36642f30bacefef405c00ba770b582eb1b2bb

olofk, 2020-06-17:
I propose we wait until the next stable verible release and then change on the Edalize side.

imphil, 2020-07-14:
@olofk there hasn't been a stable release of Verible ever, they only provide nightlies with every commit. Given that, and its limited user base, I'd propose to simply rename the tools.

olofk, 2020-08-18:
Aahh!! What the hell is wrong with all these open source EDA tool developers that just can't take an hour to put out a god damn release. Oh well, I merged the change now, so closing

commit e2d87d29 (2020-08-18): Rename Verible lint and format binary names

This renames the Verible binary names as shown below, since the old names
will eventually be deprecated.

verilog_format -> verible-verilog-format
verilog_lint   -> verible-verilog-lint

Fixes #167
closed 2020-08-18 by olofk