olofk/edalize #168: Provide verbose output option

nevercast, 2020-06-23:
When trying to debug a nextpnr issue with fusesoc and edalize, I was unable to remove the -q from the nextpnr-ice40 invocation with --verbose on the fusesoc side. 

It would be nice if edalize could be passed a verbose option, and that apply to the backends where applicable. My work around for now is to just run the command manually. 

Seems that the -q option is forced here: https://github.com/olofk/edalize/blob/79050a966bcb377f4a33a699f072bc38f0d39922/edalize/templates/icestorm/icestorm-makefile.j2#L23

imphil, 2020-06-23:
It's always easier to add options through fusesoc than to remove them; so it's a good idea to have only the absolute minimum set of options forced in the backend. `-q` isn't one of them.

Would you mind opening a PR removing the `-q` option?

olofk, 2020-11-17:
This should be fixed now. Thanks for the report and fix

closed 2020-11-17 by olofk