olofk/edalize #207: Tool installation path

qarlosalberto, 2021-01-10:
Is there a way to configure the tool installation path?

Thank you.

imphil, 2021-01-10:
What do you mean by "tool installation path"? Can you describe a bit more what problem you're trying to solve, e.g. what commands you're calling, etc.?

qarlosalberto, 2021-01-10:
I want to simulate a design with GHDL. But GHDL isn't in the system path

imphil, 2021-01-10:
In that case modify the PATH variable to add the GHDL binary directory to it; you can do so either globally, or just for your current session. Please ask Google for more help here, depending on your operating system, etc. the answer will be different.

closed 2021-01-10 by imphil

qarlosalberto, 2021-01-11:
I'm sorry for the short explanation :/

I'm adding Edalize support to TerosHDL (https://terostechnology.github.io/terosHDLdoc/). Basically I am creating a graphical interface for Edalize in VSCode:

[image]

For the user the call to Edalize is transparent. Sometimes the user doesn't have permission to modify the system path. So I have created a field where the user can set the simulator path.

I think that it would be really nice if Edalize API support a custom simulator path.

imphil, 2021-01-11:
You don't need to change the PATH variable system-wide. You only need to change the "PATH" environment variable when calling edalize, it's then "inherited" down to all called tools. So take the directory you get in the "GHDL installation path" field, and set PATH before calling edalize. If you're writing Python, you'd pass the `env` keyword to `subprocess.run()`. Depending on how your VS Code plugins calls edalize, look for something similar on probably the JS library you're using to call external programs.

qarlosalberto, 2021-01-11:
Ok! Thank you :)