olofk/edalize #209: Xcelium generated Makefile needs top level module after defparam option

seandextercrevinn, 2021-01-18:
Steps to recreate:
- Follow steps on https://github.com/lowRISC/ibex/tree/master/examples/simple_system to set up tools necessary to simulate the Ibex simple system example (i.e. Prerequisites section).
- Build the software test "hello_test" to create the hello_test.vmem required for basic simulation.
- Make the following local edit in shared/rtl/sim/simulator_ctrl.sv @ Line 59
```
-  always_ff @(posedge clk_i or negedge rst_ni) begin
+  always @(posedge clk_i or negedge rst_ni) begin
```
- Build the simulation to prepare for running with Xcelium. Use the following command from the top of the repo:
```
fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --SRAMInitFile=\\\"${PWD}/examples/sw/simple_system/hello_test/hello_test.vmem\\\"
```
- Then run the simulation with:
```
make -C build/lowrisc_ibex_ibex_simple_system_0/sim-xcelium run
```

I get the following xmelab warnings:
```
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference RV32E - DEFPARAM RV32E=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference ICache - DEFPARAM ICache=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference ICacheECC - DEFPARAM ICacheECC=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference BranchTargetALU - DEFPARAM BranchTargetALU=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference WritebackStageU - DEFPARAM WritebackStage=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference SecureIbext - DEFPARAM SecureIbex=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference BranchPredictor - DEFPARAM BranchPredictor=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference PMPEnabled - DEFPARAM PMPEnable=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference PMPGranularityr - DEFPARAM PMPGranularity=0.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference PMPNumRegionsy - DEFPARAM PMPNumRegions=4.
xmelab: *W,DEFHRF: Failed to resolve hierarchical reference SRAMInitFiles - DEFPARAM SRAMInitFile="${PWD}/examples/sw/simple_system/hello_test/hello_test.vmem"
```

I think this happens because when using the -defparam option for xrun, it needs the top level module name prefixed to the parameter. This means making the following change to the build/lowrisc_ibex_ibex_simple_system_0/Makefile generated by Edalize, at line 29:
```
-  EXTRA_OPTIONS ?= $(XRUN_OPTIONS) $(if $(XMSIM_OPTIONS),-xmsimargs '$(XMSIM_OPTIONS)',) $(addprefix -defparam ,$(PARAMETERS)) $(addprefix +,$(PLUSARGS))
+  EXTRA_OPTIONS ?= $(XRUN_OPTIONS) $(if $(XMSIM_OPTIONS),-xmsimargs '$(XMSIM_OPTIONS)',) $(addprefix -defparam $(TOPLEVEL).,$(PARAMETERS)) $(addprefix +,$(PLUSARGS))
```

I think this change can be made in https://github.com/olofk/edalize/blob/master/edalize/xcelium.py

Please let me know what you think.

olofk, 2021-06-02:
Sorry. I completely missed this. It sounds like you have found a good fix. Icarus has the same behaviour actually.

Theoretically this can be a problem with designs that have multiple top levels, but those are pretty rare so I think we can ignore those for now

PeterCDMcLean, 2021-06-18:
> but those are pretty rare so I think we can ignore those for now
Unfortunately not, I any simulation with Xilinx IP you must declare glbl a second top level to the simulation to have Xilinx IP behave properly.
see this [example](https://www.aldec.com/en/support/resources/documentation/faq/1172)

I'm piggy backing on this very similar issue because I need support for multiple top levels passed to xrun:

`XRUN_CALL = $(XRUN) -q -f edalize_main.f $(addprefix -pli ,$(VPI_MODULES)) $(EXTRA_OPTIONS) -top $(TOPLEVEL)`

needs to be:

`XRUN_CALL = $(XRUN) -q -f edalize_main.f $(addprefix -pli ,$(VPI_MODULES)) $(EXTRA_OPTIONS) $(addprefix -top ,$(TOPLEVEL))`

olofk, 2021-06-21:
Thanks for chiming in @PeterCDMcLean . It's good to know that the code for the current support for multiple toplevels at least isn't unused :)

I see three ways of fixing this.
1. Prefix all parameters with the _first_ toplevel. While not technically correct, it would probably work well for the known cases since we probably don't want to send the parameters to e.g. `glbl.v`
2. Add support for pinning parameters to a certain toplevel. I think this might need some larger changes though.
3. Send all parameters to all toplevels. I'm kind of leaning towards this solution. There are some corner cases where this won't work but realistically it would probably work just fine.

So, option 3 would be my suggestion but I need help with implementation and/or testing since I don't have a xcelium license

PeterCDMcLean, 2021-06-21:
I will go try to reproduce the OP issue

Can the user define the parameters in the core file?

```

parameters:
  ibex_simple_system/RV32E:
    datatype: int
    paramtype: vlogparam
    default: 0
    description: "E"

```

I think the only issue would be hierarchical path separator? As in top.sub
vs. top/sub

Maybe parameters could have an optional attibute to specify the prefix

```

parameters:
  RV32E:
    datatype: int
    paramtype: vlogparam
    default: 0
    description: "E"
    path: ibex_simple_system

```