olofk/edalize #214: Support collecting resource statistics from SymbiFlow related tools

mithro, 2021-01-27:
@olofk recently posted https://twitter.com/OlofKindgren/status/1354178191431245824?s=20
> I finally got around to merge a really cool Edalize feature. If you install edalize with the optional reporting module (e.g. `pip install --user -e ".[reporting]"`) you will get access to really powerful resource and timing statistics through Edalize. Check it out!
> [image]

It would be good to make sure this works with the SymbiFlow tooling including;
 - Yosys --> nextpnr
 - Yosys --> vpr

mithro, 2021-01-27:
FYI - @acomodi / @kgugala

olofk, 2021-01-28:
Yes! Definitely. Would love to have that. I'm hoping someone might be able to help out here. Happy to do it myself but I just have too much on my plate right now

syed-ahmed, 2021-02-10:
I am willing to take this up! We have a use case for edalize and this feature!

To give a little more context, we are currently working on building a partial reconfiguration framework with the symbiflow backend. An example design on a defined rrgraph has a Makefile like [this](https://gist.github.com/syed-ahmed/c4e64bc4d5e58625418cbd25567403f6). I'm currently working on generating this Makefile with edalize. In addition, the reporting feature of edalize would enable us to do autotuning on several Yosys, VTR, etc. parameters.