olofk/edalize #218: prefix run commands

rasmus-madsen, 2021-02-10:
in a corporate flow, the access to license servers is often restricted to a pool of runtime servers.
to run a simulation on one of the 3 main EDA tools you would need to submit your job to that pool.

in the current setup prefixing a command is not possible as far as I can tell.

examples:
xcelium - using NC
```xrun <runtime_options> ```
should be prefix with nc run [options], as 
``` nc run -C ncsim -Il xrun <runtime_options>```
or LSF 
```bsub [options] cxrun<runtime_options>```

similar for other LSF like systems

olofk, 2021-02-28:
ok, this is another reason to add the prefix support that I suggested in #221 . My immediate thought is that this would be something that is configure per machine, which means it could go into an edalize.conf. Would a static list of mappings between each tool (e.g. xrun, vivado, ecppack, vsim..) and a prefix be enough, or do we need to be more clever? (Please say no)

imphil, 2021-03-01:
How about going even simpler: Make it one single prefix. If users want something clever, they can have an intermediate script that does exactly what they want (e.g. dispatch Vivado jobs to another lsf job queue than ncsim jobs, for example).

olofk, 2021-03-12:
Yes! Let's do this

olofk, 2021-03-25:
I added support for running an external launcher in a couple of backends yesterday, but not xcelium.
However! I just realized that I think you can already do this

Try running `XRUN="nc run -C ncsim -Il xrun" fusesoc run ...` and see if that works

rasmus-madsen, 2021-04-09:
@olofk in cshell I have to set the environment variable prior.
which works.
I don't think this is an optimal solution. I prefer having an option I can set in a core file!