olofk/edalize #219: Testing the tool with VCS, error encountered

mahmoodulhassan-lm, 2021-02-15:
When I tried to simulate the blinky.v design by putting the code in a python script:
```
from edalize import *

import os

work_root = 'build'

files = [
  {'name' : os.path.relpath('blinky.v', work_root),
   'file_type' : 'verilogSource'},
  {'name' : os.path.relpath('blinky_tb.v', work_root),
   'file_type' : 'verilogSource'},
  {'name' : os.path.relpath('vlog_tb_utils.v', work_root),
   'file_type' : 'verilogSource'}
]

parameters = {'clk_freq_hz' : {'datatype' : 'int', 'default' : 1000, 'paramtype' : 'vlogparam'},
              'vcd' : {'datatype' : 'bool', 'paramtype' : 'plusarg'}}

tool = 'vcs'

edam = {
  'files'        : files,
  'name'         : 'blinky_project',
  'parameters'   : parameters,
  'toplevel'     : 'blinky_tb'
}

backend = get_edatool(tool)(edam=edam,
                            work_root=work_root)

os.makedirs(work_root)
backend.configure()

backend.build()

args = {'vcd' : True}
backend.run(args)
```
I got the following error:
```
Traceback (most recent call last):
  File "edalize_test.py", line 32, in <module>
    backend.configure()
  File "/home/mahmood.hassan/Edalize/edalize/edalize/edatool.py", line 169, in configure
    self.configure_main()
  File "/home/mahmood.hassan/Edalize/edalize/edalize/vcs.py", line 83, in configure_main
    self.render_template('Makefile.j2', os.path.join(self.work_root, 'Makefile'), template_vars)
  File "/home/mahmood.hassan/Edalize/edalize/edalize/edatool.py", line 315, in render_template
    with open(file_path, 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'build/build/Makefile'
```
It is observed in the last line that Edalize trying to make build directory in another build which should not be the case.

Is there any complete document consisting of configurations for each EDA tool?

olofk, 2021-02-28:
Aha! I found (and fixed) the bug. Thanks for reporting. I pushed a fix now. Please let me know if that works.

Tool options for each backend can be found here https://edalize.readthedocs.io/en/latest/edam/api.html#tool-options but there's not much more than that for the individual tools

olofk, 2021-06-02:
I believe this should be fixed now. Please reopen otherwise

closed 2021-06-02 by olofk