olofk/edalize #229: GHDL Backend Support for Toplevel in Library

GCHQDeveloper560, 2021-04-08:
VHDL designs that heavily use libraries will often specify the top level in FuseSoC with a dot notation, such as "my_lib.top_design". A number of tools including ModelSim and Vivado seem to support this common notation, but it's [not supported by GHDL](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#elaboration-e):

> Name of the units must be a simple name, without any dot.

GHDL instead suggests using something like
```
ghdl -m --work=my_lib --workdir=./my_lib top_design
```

Given use of this format with other tools it feels like it would be worth what should be a fairly simple workaround in the GHDL backend. Is this of interest, or do I need to see about getting this supported in GHDL?

olofk, 2021-04-08:
Having ghdl support the dot notation would be preferred in the long run, but won't help current users, so I think it's a good idea to split up the toplevel name like this for the ghdl backend

GCHQDeveloper560, 2021-04-08:
A recent commit indicates @m-kru is planning more improvements to the GHDL backend, so perhaps this is already being addressed. I at least want to avoid conflicts!

m-kru, 2021-04-09:
@tgingold I guess GHDL is not going to support the dot notation?

tgingold, 2021-04-09:
You mean something like `ghdl -e work.top` ?

That could be implemented.

m-kru, 2021-04-09:
@GCHQDeveloper560

GCHQDeveloper560, 2021-04-09:
It would be great to have that in future GHDL revisions! I [filed a GHDL issue](https://github.com/ghdl/ghdl/issues/1718) with the request.

m-kru, 2021-04-09:
I do not know how hard it would be to add it to the GHDL, but such approach would be preferable as it would make GHDL more "unified" with other EDA tools and would allow to avoid boilerplate in the Edalize.

commit 40ed948d (2021-04-09): Add support for top level that includes a library specification

Fixes #229.

Many tools allow the top level unit in a non-default library to be
specified using the syntax my_lib.my_unit. GHDL doesn't and instead
requires the library to be specified separately with the ``--work`` and
``--workdir`` options. This converts the common dot notation to GHDLs
syntax.

This change also reduces the number of GHDL commands by combining all
source files for a library into a single command rather than using a
GHDL command for every file.
closed 2021-04-09 by olofk