olofk/edalize #230: Add Support for VHDL to Yosys and SymbiYosys Backends Via the Yosys GHDL Plugin

GCHQDeveloper560, 2021-04-08:
The [Yosys GHDL plugin](https://github.com/ghdl/ghdl-yosys-plugin) is still listed as experimental but is already very useful. For formal verification GHDL's support for PSL seems more extensive than the Yosys Verilog assertion support. Currently the Yosys and SymbiYosys backends only support Verilog files, but it would be great to add support for the plugin if we can tolerate potential changes due to its early status.

It appears the plugin currently only offers a subset of the GHDL analysis features. More complex usage seems to assume that GHDL is used to analyse the files, and then the plugin is used inside Yosys to perform elaboration. This may mean it makes sense to add support for using GHDL as a sub-tool as is already done for Yosys. I don't know whether the ``-i`` flow used by the GHDL backend works, or if the ``-a`` flow is required.

olofk, 2021-04-08:
I'm all for this, and I see it as part of a general effort to add support for adding different parser frontends to Edalize. Other tools in this family include surelog and sv2v. The status right now is that I'm trying to figure out whether to do this as a stand-alone effort (like the yosys support) or do this as part of the transition to the next-generation Edalize https://github.com/olofk/edalize/wiki/Edalize-(Slight-return)

antonblanchard, 2021-08-10:
We'd also be interested in this for [Microwatt](https://github.com/antonblanchard/microwatt). Right now we use fusesoc for synthesis with the proprietary tools, and Makefiles for synthesis with yosys. Being able to use fusesoc everywhere would be very nice.

olofk, 2021-11-17:
With the new Edalize architecture starting to fall in place I think it's a good opportunity to revisit this. I added support for using sv2v as a frontend yesterday to enable sv support for yosys and I would like to do the same with ghdl for vhdl. I'm not sure I understand how it all fits together though. Is there both a ghdl yosys plugin and a way to run ghdl standalone to output verilog? If so, the latter has the advantage that we can put it in front of other tools as well (e.g. verilator)

nobodywasishere, 2023-01-07:
Yes, both ghdl-yosys-plugin and ghdl synth exist, the latter able to be used standalone.

ru551n, 2024-06-24:
What is the status of this issue? I can see it has been dormant for some time and I also very interested in this.

olofk, 2024-06-25:
I believe it should be doable now if you use the Edalize flow API (some info about how to choose the flow API here https://fusesoc.readthedocs.io/en/latest/user/build_system/eda_flows.html) and then set `frontends: [ghdl]` in your `flow_options`. Please let me know if that works for you.