olofk/edalize #241: Extra Vivado options

gmartin102, 2021-05-18:
This maybe not the correct place for this issue/question, but maybe someone can point me in the correct direction.  I am working with openhwgroup on an SoC that is using fusesoc to build fpga based emulation hardware.  All is working ok, except the vivado tools are using the flatten-hierarchy=rebuilt oprion and I would liek to have it be flatten_hierarchy=none.  I've tried a couple of  thibngs but no luck on getting it to actually do what i want.  how should I add tcl command options for vivado to use in the sysnthesis?  the following is a snip from my core file.

```
 nexys-a7-100t:
    <<: *default
    default_tool: vivado
    description: Digilent Nexys-A7-100T Board
    parameters:
    - PULP_FPGA_EMUL=true
    - flatten_hierarchy=none
    filesets_append:
    - nexys-cfg
    - nexys-xdc
    tools:
      vivado:
        part: xc7a100tcsg324-1
        flatten_hierarchy: none
    toplevel: xilinx_core_v_mcu
```

olofk, 2021-05-20:
You have come to the right place

As a philosophy, Edalize tries to expose as few tool-specific options as possible since there are far too many options to list them all. For custom options like this, you will need to find the correct option in the Vivado TCL API, write it in a tcl file, and add it to a fileset with file_type set to `tclSource` This file will then get sourced by Edalize when the project is created

olofk, 2021-05-20:
Hint, I often find that the easiest way to figure out what command to use in cases like this is to start the Vivado GUI on a synthesized design, change the setting with the GUI and look at the TCL console to see what command was run. Just tried it and it seems you need something like `set_property STEPS.SYNTH_DESIGN.ARGS.FLATTEN_HIERARCHY none [get_runs synth_1]`

gmartin102, 2021-05-20:
Thanks for the reply.  I did exactly what you suggested and it works like a charm !  thanks again

closed 2021-05-20 by gmartin102