olofk/edalize #244: Tool path

qarlosalberto, 2021-06-16:
It would nice to support:

- Tool in system path: get the ModelSim path from vsim binary.
- Pass the tool path from Python. Sometimes the user isn't root and he can't add the tool to the system path.

imphil, 2021-06-16:
> Tool in system path: get the ModelSim path from vsim binary.

Can you elaborate on that a bit and describe your use case? What is the "ModelSim path"?

> Pass the tool path from Python. Sometimes the user isn't root and he can't add the tool to the system path.

Even as user, you can control the PATH environment variable. Does that not work for you?

qarlosalberto, 2021-06-16:
- Currently MODEL_TECH is required, even though ModelSim is on the system path. Maybe Edalize could use (as cocotb): 

```
from distutils.spawn import find_executable
find_executable("vsim")
```
- Yes, it works for me. But I'm integrating Edalize in other tool and I'm thinking in other users. I think that the VUnit way to set the tool path is very good: https://vunit.github.io/cli.html#simulator-selection

imphil, 2021-06-16:
On MODEL_TECH: I agree getting rid of it would be a good thing, and detecting it from the vsim path sounds good.

On providing tool-specific path overrides: I disagree that having individual options for tool paths is desirable. There are dozens of tools called by edalize. There's a universal way to set the path to those tools, and that's the PATH environment variable. That variable is easily changed, even from wrapper scripts which pass on a modified environment, by non-privileged users, etc. I don't see any benefit of duplicating this functionality inside edalize.

qarlosalberto, 2021-06-16:
Yes, you are right. I could add the tool path from my Python script.