olofk/edalize #252: Documentation for edalize isn't auto-generated

imphil, 2021-07-05:
edalize describes backend options as part of the backend class. This information is used to generate backend documentation in FuseSoC's documentation. 

However, for edalize's own documentation, we have a hardcoded version of the documentation at https://github.com/olofk/edalize/blob/master/doc/edam/api.rst, which is outdated/not in sync with the other documentation.

@olofk Should we switch the edalize one to the auto-generated docs as well? Or was there something else blocking/motivating that?

olofk, 2021-07-05:
We should definitely fix this. I was just made aware of it last week or so when @qarlosalberto pointed out that some options weren't documented. Ok, made aware probably isn't the right word since it's my fault that it is like this but I had completely forgotten about it and thought it was automatically updated just like FuseSoC. I suppose the easiest way would be to just steal some of the code from FuseSoC that creates the docs. There's another thing as well that I noticed. Some of the data types used in the Edalize tool options aren't valid. We need to fix that as well

olofk, 2024-01-11:
Hey! I actually fixed this a couple of weeks ago. Woohoo!!

closed 2024-01-11 by olofk