olofk/edalize #253: How to determine the incdir of. V file？

brodiezhu, 2021-07-05:
How to determine the incdir of. V file？

imphil, 2021-07-05:
Can you elaborate a bit more on what you're trying to achieve?

brodiezhu, 2021-07-06:
Source file include other source files, how do we add the path of the include file？

imphil, 2021-07-06:
If you set `is_include_file` for a file, the path of the file is passed to the simulator or synthesis tool as include path. See https://edalize.readthedocs.io/en/latest/edam/api.html.

closed 2021-07-06 by imphil

brodiezhu, 2021-07-06:
I have another question, how to set + define + macro = vaule, can you tell me?

imphil, 2021-07-06:
Please again have a look at the API documentation at https://edalize.readthedocs.io/en/latest/edam/api.html#parameter.

brodiezhu, 2021-07-07:
Thank you very much

brodiezhu, 2021-07-07:
I also have a question, if I want to add a new EDA tool to simulate the case, is this behavior feasible?