olofk/edalize #258: Install failure; dependencies missing, consider gracefully checking versions

gojimmypi, 2021-07-17:
I tried installing `edalize` via the [instructions](https://github.com/olofk/edalize#install-it) from my fresh fork but I saw an error that looked dependency-related:  Note there's a `ContextualVersionConflict` warning about `Jinja2`. This is on WSL, so admittedly perhaps not the ideal environment.

```
gojimmypi:~
$ cd /mnt/c/workspace
gojimmypi:/mnt/c/workspace
$ git clone https://github.com/gojimmypi/edalize.git
Cloning into 'edalize'...
remote: Enumerating objects: 2848, done.
remote: Counting objects: 100% (500/500), done.
remote: Compressing objects: 100% (232/232), done.
remote: Total 2848 (delta 303), reused 422 (delta 257), pack-reused 2348
Receiving objects: 100% (2848/2848), 781.26 KiB | 829.00 KiB/s, done.
Resolving deltas: 100% (1764/1764), done.
gojimmypi:/mnt/c/workspace
$ cd edalize
gojimmypi:/mnt/c/workspace/edalize
$ python3 -m pip install -e .
Obtaining file:///mnt/c/workspace/edalize
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 1025, in check_if_exists
    self.satisfied_by = pkg_resources.get_distribution(str(no_marker))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 472, in get_distribution
    dist = get_provider(dist)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 344, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 892, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 783, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (Jinja2 2.10 (/usr/lib/python3/dist-packages), Requirement.parse('Jinja2>=2.11.3'), {'edalize'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 521, in _prepare_file
    req_to_install.check_if_exists()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 1036, in check_if_exists
    self.req.name
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 472, in get_distribution
    dist = get_provider(dist)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 344, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 892, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 783, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (Jinja2 2.10 (/usr/lib/python3/dist-packages), Requirement.parse('Jinja2>=2.11.3'), {'edalize'})
gojimmypi:/mnt/c/workspace/edalize
$ python3 --version
Python 3.6.9
gojimmypi:/mnt/c/workspace/edalize
$ python --version
Python 2.7.17
gojimmypi:/mnt/c/workspace/edalize
```

Here's the result of `pip freeze`
```
$ pip freeze
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
[... 148 lines cut ...]
Collecting Jinja2
  Downloading https://files.pythonhosted.org/packages/80/21/ae597efc7ed8caaa43fb35062288baaf99a7d43ff0cf66452ddf47604ee6/Jinja2-3.0.1-py3-none-any.whl (133kB)
    100% |████████████████████████████████| 143kB 340kB/s
Collecting MarkupSafe>=2.0 (from Jinja2)
  Downloading https://files.pythonhosted.org/packages/fc/d6/57f9a97e56447a1e340f8574836d3b636e2c14de304943836bd645fa9c7e/MarkupSafe-2.0.1-cp36-cp36m-manylinux1_x86_64.whl
Installing collected packages: MarkupSafe, Jinja2
Successfully installed Jinja2-3.0.1 MarkupSafe-2.0.1
```
It looks like the next attempt was successful:
```
$ python3 -m pip install -e .
Obtaining file:///mnt/c/workspace/edalize
Collecting Jinja2>=2.11.3 (from edalize==0.2.5)
  Using cached https://files.pythonhosted.org/packages/80/21/ae597efc7ed8caaa43fb35062288baaf99a7d43ff0cf66452ddf47604ee6/Jinja2-3.0.1-py3-none-any.whl
Collecting MarkupSafe>=2.0 (from Jinja2>=2.11.3->edalize==0.2.5)
  Using cached https://files.pythonhosted.org/packages/fc/d6/57f9a97e56447a1e340f8574836d3b636e2c14de304943836bd645fa9c7e/MarkupSafe-2.0.1-cp36-cp36m-manylinux1_x86_64.whl
Installing collected packages: MarkupSafe, Jinja2, edalize
  Found existing installation: edalize 0.2.5
    Can't uninstall 'edalize'. No files were found to uninstall.
  Running setup.py develop for edalize
Successfully installed Jinja2-3.0.1 MarkupSafe-2.0.1 edalize
gojimmypi:/mnt/c/workspace/edalize
```

Perhaps this is a feature request to more gracefully check dependency versions and optionally prompt to automatically update them?

mithro, 2021-07-17:
The pip error message indicated that your version of Jinja2 was to old as I would expect.

If you would like pip to automatically upgrade required dependencies then you need to give it a `--upgrade`. See https://pip.pypa.io/en/stable/cli/pip_install/#overview ;
> Note that `pip install` prefers to leave the installed version as-is unless `--upgrade` is specified.

gojimmypi, 2021-07-17:
there's an [install_requires](https://github.com/olofk/edalize/blob/a09c806c190145133b98b0a5f79afedfa9ebe986/setup.py#L53) section that _checks_... do you think it would be good to prompt the user as to exactly what should be done? The error is a bit long and intimidating. Perhaps give the option to install Edalize with a similar `--upgrade` option that also does the pip upgrade? (btw: note that my `pip` needed to be specified as `pip3`)

imphil, 2021-07-19:
pip typically does not change system-wide dependencies, and Jinja is installed in that way on your machine. 

The error message isn't great and it's not how it normally looks. But either way, this is something edalize cannot do anything about, since we use the standard ways to install packages. Most likely something is wrong in your Python installation. I'd recommend asking in a general Python forum, since I don't really have more insight into what's going on here.