olofk/edalize #263: Add VTR backend

jgoeders, 2021-08-23:
I'm going to work on running VTR from edalize, and will track progress here.

mithro, 2021-08-23:
@jgoeders - I'm pretty sure Antmicro has already added this work. @kgugala - Who was working on this?

mithro, 2021-08-23:
See https://github.com/olofk/edalize/pulls?q=is%3Apr+vpr+is%3Aclosed

jgoeders, 2021-08-23:
OK, those look like like nextpnr and Symbiflow backends.  I thought in the VTR meeting we had discussed creating an Edalize backend specifically for VTR.  Am I misunderstanding?

mithro, 2021-08-23:
The SymbiFlow backend uses VPR, so I think the work is just making sure the SymbiFlow backend is usable for the VtR use cases?

jgoeders, 2021-09-15:
Basic backend added in #271 .

Will update original post with plan for future functionality.