olofk/edalize #284: Typo for xsim.py

MikeOpenHWGroup, 2021-11-02:
OK, this is being _really_ picky, but I believe the long version of the "snapshot" command-line argument should be prefixed with two dashes.  That is,
```
--snapshot $(TARGET)
```
not
```
-snapshot $(TARGET)
```
See line [23 of xsim.py](https://github.com/olofk/edalize/blob/master/edalize/xsim.py#L23).   This is a low priority issue since it seems that xsim accepts both forms.

imphil, 2021-11-02:
Really? The Xilinx manual seems to agree suggest `-snapshot`.

[image]

MikeOpenHWGroup, 2021-11-02:
Ha!  You are right.  But so am I!!  Well, not really.  Try `xelab -h` and you get:
```
Usage: xelab [options] [libname.]unitname...
(Switches with double dash '--' can also be used with a single dash '-')

Vivado Simulator xelab options:
******************************:
  -a [ --standalone ]              Generates a standalone non-interactive 
                                   simulation executable that performs run-all.
[snip]
 -s [ --snapshot ] arg            Specify the name of design snapshot
[snip]
```

I failed to notice that line at the top: `(Switches with double dash '--' can also be used with a single dash '-')`.  That's just bad user documentation, but you are right that it isn't an edalize bug.

closed 2021-11-02 by MikeOpenHWGroup

imphil, 2021-11-02:
Beautiful consistency :-) Thanks for the bug report nonetheless!