olofk/edalize #291: Missing apicula/apicula-makefile.j2 for Gowin targets

timvgso, 2021-12-17:
I tried the *LED  to believe* tutorial in [https://github.com/fusesoc/blinky](https://github.com/fusesoc/blinky) for a Sipeed Tang Nano board with a fresh install of FuseSoc, Yosys, and Apicula with Python 3.9.5 on Linux Mint x86_64.

I installed the stand-alone `blinky` core to get support for the `tang_nano` target, with:
`fusesoc library add blinky https://github.com/fusesoc/blinky`

Building the project with:
`fusesoc run --target=tang_nano fusesoc:utils:blinky`

failed with output of:
```
INFO: Preparing fusesoc:utils:blinky:1.0
INFO: Setting up project
INFO: Setting up project
[shortened ...]
  File "/media/timv/REDCAT1/files/fusesoc/.env/lib/python3.9/site-packages/jinja2/loaders.py", line 335, in get_source
    raise TemplateNotFound(template)
jinja2.exceptions.TemplateNotFound: apicula/apicula-makefile.j2
```

With some web searching, I found a page at [https://github.com/E4tHam/tangnano_example](https://github.com/E4tHam/tangnano_example) that said:
`Note: As of 9/12/21, Edalize has a bug with the Apicula toolchain. To fix it, after installing FuseSoC, manually add this Makefile template to "${YOUR_EDALIZE_INSTALL_DIR}/templates/apicula/apicula-makefile.j2".`

The Makefile template is linked at [https://github.com/infphyny/edalize-1/blob/master/edalize/templates/apicula/apicula-makefile.j2](https://github.com/infphyny/edalize-1/blob/master/edalize/templates/apicula/apicula-makefile.j2). The file `apicula-makefile.j2` and the containing `apicula` directory do not currently exist in [https://github.com/olofk/edalize/tree/master/edalize/templates](https://github.com/olofk/edalize/tree/master/edalize/templates) and were not installed in the `lib/python3.9/site-packages/edalize/templates` directory on my local machine.

After creating the directory `apicula` and manually installing the file `apicula-makefile.j2`, I was able to successfully build the `blinky` example with the same command.

This is all very new to me and I don't know anything about Edalize or Jinja but I'll be happy to help if there's anything more I can do.

E4tHam, 2021-12-17:
Thanks for making this issue! I forgot to make one when I discovered the problem.

olofk, 2022-01-07:
Thank you for the bug report. Hadn't seen @E4tHam's page before either. Very cool!

So, yes, I that there is a bug in the latest stable version of edalize. The apicula backend has been rewritten since then, so the development version should work until a new release is made. I've been planning to put out a new release for a while so I might just take the opportunity to do that right away actually. Will report back after that

olofk, 2022-01-07:
Could you please check if this is still an issue with version 0.3.0 of Edalize?

timvgso, 2022-01-07:
Sure I'll check it out asap. Thanks for the fix!

olofk, 2022-01-12:
@timvgso @E4tHam Just wanted to check if anyone has had an opportunity to check if Edalize 0.3.0 fixes this (no rush!)

timvgso, 2022-01-13:
> @timvgso @E4tHam Just wanted to check if anyone has had an opportunity to check if Edalize 0.3.0 fixes this (no rush!)

Yes, I just tested it. Pip in a fresh virtualenv installs edalize-0.3.0, which builds blinky for tang_nano  successfully. Thanks for the fix!

I don't see an `apicula` directory or an `apicula-makefile.j2` file installed, but I'm supposing that it works differently now with the rewritten Apicula backend and doesn't have that file. (Putting that in for the possible benefit of someone who might find this page down the road and think it's still a problem.)

olofk, 2022-01-13:
Thanks for testing and reporting! And yes, there's no makefile template anymore. The makefile is created programmatically (and in the future there will be an option to generate ninja files, sh/bat files or launch the tools directly from python to avoid Makefile issues that plague windows users).

Closing now

closed 2022-01-13 by olofk