olofk/edalize #302: Improve SymbiFlow backend

olofk, 2022-01-28:
The SymbiFlow backend is currently a mixed bag with a yosys+nextpnr-centric flow and a yosys+vpr-centric flow with the latter being called through the symbiflow_* shell script wrappers. The drawback of going through those shell script wrappers is that some functionality (such as settings defines, parameters sourcing TCL scripts etc) are not exposed in the wrappers. The idea now is to call the underlying tools directly starting with replacing symbiflow_synth. A small first step is done through https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2371 and this is done partly to better support https://github.com/olofk/edalize/pull/299

[image]
h

carlosedp, 2022-04-28:
I've also started implementing "some" symbiflow for Kintex devices as a Prj-Xray backend that messes things up a bit too.

Maybe it's needed to take a broad look into the ecosystem to avoid big refactors in the future again. Maybe homogenize under the f4pga umbrella is a good idea.

https://github.com/olofk/edalize/pull/308

carlosedp, 2022-04-28:
Also I've raised some points about the Quicklogic flow for Symbiflow here: https://gist.github.com/carlosedp/ed17d934c94338cbb2f9b29dd076bf1a

There are still some open points that needs to be addressed.

jgoeders, 2022-05-31:
@olofk See above.  Can you point us to an example of using ESR?  Thanks!

olofk, 2022-05-31:
Apologies. Been out of the loop for a bit. There are a few backends that have been ported over to the new flow. [This one](https://github.com/olofk/edalize/blob/master/edalize/flows/icestorm.py) might be the best example. And to answer your other question, a makefile is still generated just like before. We just do it more programatically now.

And warmly welcom @Pocketkid2 and @schafernc. Don't hesitate to reach out if you have any questions