olofk/edalize #315: vopt_options missing in modelsim.py

oholimoli, 2022-04-22:
Hi, I have question if there is a reason why there are no `vopt_options` for Modelsim available? =>
https://github.com/olofk/edalize/blob/master/edalize/modelsim.py#L78

Would if be straight forward to add `vopt_options` hasn't this option been added for a specific reason?

imphil, 2022-04-24:
vopt is not called by edalize, as far as I see it -- therefore no options are needed. Or do you see calls to vopt that I missed (entirely possible!)?

oholimoli, 2022-04-25:
Would it be possible or make sense at all to call vopt with edalize (in case there are vopt options)? What would be the required steps to add the vopt part to edalize?
Sorry I'm not a Modelsim expert. I'm trying to write a core file for fusesoc for an existing design where vopt is used and would like to add all vopt options the same way to the core as it is done with vsim/vcom/vlog.

shareefj, 2022-05-05:
The Edalize modelsim flow uses the two step flow and not the three step.  You need to pass the vopt options you want in vsim_options.

The usual vsim command line would be something like: `vsim <vsim_options> -voptargs=<vopt_options>`.  Take a look in the manual and search for `voptargs`.