olofk/edalize #317: Vivado method missing for programming FPGA (with fusesoc)

patrickerich, 2022-05-12:
In previous versions of edalize there was a method available for running the "pgm" target of the generated Vivado Makefile.
This was very useful with fusesoc (during the "run" phase) and seems to currently be missing.

Will this be added back in?
Something similar to:
```self._run_tool("make", ["pgm"])```

imphil, 2022-05-18:
The `fusesoc pgm` command-line argument was deprecated for a long time and has now been removed from FuseSoC. However, the functionality to program a device is still there in the latest FuseSoC version, you can call it through `fusesoc run --target=synth --run`.

See also https://github.com/olofk/fusesoc/commit/dd16521ebb55d6f1ea12649e2d8fe8ada083216e

patrickerich, 2022-05-18:
Sorry for the misunderstanding.
I have never used ```fusesoc pgm```.
The ```fusesoc run --target=synth --run``` is exactly what is no longer working.

The file edalize/vivado.py implements the ```class Vivado(Edatool)``` which instantiates ```self.vivado = Vivado_underlying(edam, work_root, verbose)``` with ```from edalize.flows.vivado import Vivado as Vivado_underlying```
The file edalize/flows/vivado.py implements the ```class Vivado(Edaflow)``` whose parent class (Edaflow) contains:
```python
    # Most flows won't have a run phase
    def run(self, args):
        pass
```

imphil, 2022-05-18:
Oh, it looks like this functionality then got lost as part of https://github.com/olofk/edalize/pull/303. That's clearly not intended.

If you have an idea for a fix please feel free to open a PR. (I won't get around to do so for the next days.)

patrickerich, 2022-05-18:
A simple fix would be to restore the original ```run_main``` method in the edalize/vivado.py file (which is still in the current pypi implementation of edalize):

```python
    def run_main(self):
        """
        Program the FPGA.

        For programming the FPGA a vivado tcl script is written that searches for the
        correct FPGA board and then downloads the bitstream. The tcl script is then
        executed in Vivado's batch mode.
        """
        if "pnr" in self.tool_options:
            if self.tool_options["pnr"] == "vivado":
                pass
            elif self.tool_options["pnr"] == "none":
                return

        self._run_tool("make", ["pgm"])
```
This might however not be in line with the more recent changes, which seem to move more towards the ```edalize.flows```.
For now, I will create a fork with this change (test it) and open a PR.

olofk, 2022-05-31:
Sorry about this. I was working on a larger refactoring and kind of ran out of time halfway through this. I would like to add that long term I want to completely rework how device programming works. The original assumption was that there would be one programming tool for each backend, but this is not how things work in practice.

patrickerich, 2022-07-06:
Thanks for merging #319.