olofk/edalize #323: F4PGA/Symbiflow Backend

Pocketkid2, 2022-06-17:
This is a project that @schafernc and I are working on this summer under the supervision of @jgoeders. This is essentially our attempt at implementing #302 to rework the F4PGA/Sybmiflow Edalize backend. 

We decided to start an entirely new module called F4PGA (to reflect the new name of the Symbiflow project) that uses the new flow method (I believe this is also called slight return). This way we can use the new Yosys and VPR tools instead of just doing everything in the Symbiflow tool as it exists currently.

Our first major goal is to make this new module function how the existing toolchain works but have it call the tools directly instead of using the included shell wrappers. After that, we can add new features as needed. Once this project becomes stable we'll consider having this PR merged.

Tasklist:

- [x] Finish barebones Yosys/VPR flow that can generate a working Basys3 bitstream (completed in #322)
- [x] Move all hard-coded values to parameters (completed in #322)
- [x] Add some basic documentation (completed in #322)
- [x] Add basic test (completed in #322)
- [ ] Parameterize VPR modification (so that the VPR module doesn't hold F4PGA specific values)
- [ ] Support SDC files
- [ ] Support PCF files
- [ ] Support Quicklogic architecture
- [ ] Support NextPNR
- [ ] More advanced tests
- [ ] More documentation with examples

mithro, 2022-07-06:
@umarcor - It would nice if most of the scripts listed under arch-defs-install went away;
[image]

umarcor, 2022-07-06:
@mithro, agree. Is it within the scope of this issue? The description mentions "shell wrappers", not "Python wrappers".

Pocketkid2, 2022-07-06:
> @mithro, agree. Is it within the scope of this issue? The description mentions "shell wrappers", not "Python wrappers".

@schafernc and I have been directed to make the Edalize backend call the tools directly without the need for additional python wrappers (because Edalize is already a python module). We are, however, using the existing wrappers as a guide to help us design the new F4PGA Edalize backend so that it works in roughly the same way.

mithro, 2022-07-06:
@umarcor - Well, I was thinking that the work making edalize call the python scripts might be better spent making the python scripts disappear...

jgoeders, 2022-07-06:
@mithro I agree.  

@Pocketkid2 There are "wrappers" that are calling the individual tools, and then there are the set of Python scripts that @mithro  highlighted that are doing extra processing along the way. 

We will start by getting rid of the wrappers, but still call those Python scripts (just to get functionality deployed faster), but then we should tackle those Python scripts as well.  We will need to look at them on a case-by-case basis to see if they should be migrated into the Edalize sub-tool, or whether it's better to make modifications to the tools themselves (eg modify VPR).  I'm not familiar with these scripts enough to know.  

@Pocketkid2 Can you add a new task item above with these scripts (with sub-bullet items for each script)?

mithro, 2022-07-06:
@jgoeders - Basically, I'm pretty sure all the functionality should end directly in the VPR tool or Yosys plugin.

jgoeders, 2022-07-06:
> @jgoeders - Basically, I'm pretty sure all the functionality should end directly in the VPR tool or Yosys plugin.

OK, thanks.  Good to know.

mithro, 2022-07-06:
@kgugala / @acomodi - Antmicro where involved in creating a lot of these "extra" Python scripts, can you help advise?

umarcor, 2022-07-06:
@Pocketkid2, there are five sets of "scripts":

- Distributed as part of the arch-defs packages (tarballs):
  - Lowest abstraction: Python scripts. Some of these do have Python dependencies, which are not explicit at the moment. See https://github.com/SymbiFlow/f4pga-arch-defs/tree/main/utils and https://github.com/SymbiFlow/f4pga-arch-defs/blob/main/requirements.txt.
  - Low abstraction: TCL scripts. I don't know whether those are plain TCL or whether they have some additional dependencies. See SymbiFlow/f4pga-arch-defs#2371.
  - Intermediate abstraction: Bash scripts. These were moved last month from arch-defs to f4pga.
- Distributed as Python package f4pga:
  - Higher abstraction: Bash scripts wrapped in Python entrypoints. For each deprecated `symbiflow_*` script, a Python function exists. The Python function is used to generate an entrypoint through `setup.py`, and it calls the corresponding shell/bash script. See https://github.com/chipsalliance/f4pga/blob/main/f4pga/wrappers/sh/__init__.py and https://github.com/chipsalliance/f4pga/blob/main/f4pga/setup.py#L53-L77.
- Highest abstraction: any workflow/pipeline management tool written in Python (Litex, f4pga, Edalize, Xeda...), or Makefiles.

Therefore, Python is both on top and below the TCL/Bash scripts. Replacing the Bash scripts with Python code is technically straightforward because those are isolated. I.e., you can fork f4pga and replace the wrapper functions with any Python code to test backwards compatibility of the Python implementation.
However, the lowest abstraction python scripts were not isolated yet. Those should be identified and moved somewhere else (VPR, Yosys, plugins, f4pga...), to decouple them from the arch-defs packages/tarballs.

Pocketkid2, 2022-07-06:
> My approach to the split_io script, which is one of those custom scripts used, was to call it from a tcl file which we tell yosys to execute. The end result is that we only call yosys once, and it will do the magic internally.

I fully support this answer, I'll add it to the list