olofk/edalize #361: support for include file in yosys

bat52, 2022-12-27:
I have been experimenting with various EDA tools including edalize with a small-but-not-too-small open source project here https://github.com/bat52/cryptech . 

This particular verilog code has some verilog files that contain ``include "something.vh" `  primitives.

When calling yosys directly, I have managed to get those to be digested by writing
`read -incdir path`
into a .ys script as you can see here
https://github.com/bat52/cryptech/blob/master/pycryptech/ecdsa256_yosys.py

However, I am unable to achieve the same when using edalize as a 'middleman'.
I have looked into yosys.py within edalize, and it seems incdirs are somehow extracted from the included files. I also see the edalize_yosys_procs.tcl template includes 
`proc set_incdirs {} {
}`
But I could not find how to actually write the incdirs inside.
You can see the current edalize/yosys code here:
https://github.com/bat52/cryptech/blob/master/pycryptech/edalize_wrapper.py

Perhaps you can suggest what would be the correct method to get this done.

olofk, 2023-02-23:
You need to set the attribute `is_include_file : true` for files that are supposed to be `include'd

bat52, 2023-02-23:
Thank you very much! It worked indeed!

closed 2023-02-23 by bat52