olofk/edalize #368: Testsuite fails intermittently

yurivict, 2023-01-08:
In most test runs many tests fail.
But every once in a while al tests succeed.
It looks like there is a dependency on the order tests are run. (PyTest runs them in parallel.)

olofk, 2023-02-23:
I have not been able to reproduced this myself. The problem might still exist, but there's not much I can do about it without more information. If you have any logs from a failed run, please add them here. Otherwise I will have to close this without action

yurivict, 2023-02-23:
For example, I got this error:
```
=============================================================================================== FAILURES ================================================================================================
_____________________________________________________________________________________________ test_modelsim _____________________________________________________________________________________________

self = <edalize.modelsim.Modelsim object at 0x91b546b50>, cmd = 'gmake', args = [], quiet = True

    def _run_tool(self, cmd, args=[], quiet=False):
        logger.debug("Running " + cmd)
        logger.debug("args  : " + " ".join(args))
    
        capture_output = quiet and not (self.verbose or self.stdout or self.stderr)
        try:
>           cp = run(
                [cmd] + args,
                cwd=self.work_root,
                stdin=subprocess.PIPE,
                stdout=self.stdout,
                stderr=self.stderr,
                capture_output=capture_output,
                check=True,
            )

edalize/edatool.py:460: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

input = None, capture_output = False, timeout = None, check = True, popenargs = (['gmake'],)
kwargs = {'cwd': '/tmp/pytest-of-yuri/pytest-70/test_modelsim0/0', 'stderr': None, 'stdin': -1, 'stdout': None}, process = <Popen: returncode: 2 args: ['gmake']>, stdout = None, stderr = None
retcode = 2

    def run(*popenargs,
            input=None, capture_output=False, timeout=None, check=False, **kwargs):
        """Run command with arguments and return a CompletedProcess instance.
    
        The returned instance will have attributes args, returncode, stdout and
        stderr. By default, stdout and stderr are not captured, and those attributes
        will be None. Pass stdout=PIPE and/or stderr=PIPE in order to capture them.
    
        If check is True and the exit code was non-zero, it raises a
        CalledProcessError. The CalledProcessError object will have the return code
        in the returncode attribute, and output & stderr attributes if those streams
        were captured.
    
        If timeout is given, and the process takes too long, a TimeoutExpired
        exception will be raised.
    
        There is an optional argument "input", allowing you to
        pass bytes or a string to the subprocess's stdin.  If you use this argument
        you may not also use the Popen constructor's "stdin" argument, as
        it will be used internally.
    
        By default, all communication is in bytes, and therefore any "input" should
        be bytes, and the stdout and stderr will be bytes. If in text mode, any
        "input" should be a string, and stdout and stderr will be strings decoded
        according to locale encoding, or by "encoding" if set. Text mode is
        triggered by setting any of text, encoding, errors or universal_newlines.
    
        The other arguments are the same as for the Popen constructor.
        """
        if input is not None:
            if kwargs.get('stdin') is not None:
                raise ValueError('stdin and input arguments may not both be used.')
            kwargs['stdin'] = PIPE
    
        if capture_output:
            if kwargs.get('stdout') is not None or kwargs.get('stderr') is not None:
                raise ValueError('stdout and stderr arguments may not be used '
                                 'with capture_output.')
            kwargs['stdout'] = PIPE
            kwargs['stderr'] = PIPE
[... 76 lines cut ...]
                capture_output=capture_output,
                check=True,
            )
        except FileNotFoundError:
            _s = "Command '{}' not found. Make sure it is in $PATH".format(cmd)
            raise RuntimeError(_s)
        except subprocess.CalledProcessError as e:
            _s = "'{}' exited with an error: {}".format(e.cmd, e.returncode)
            logger.debug(_s)
    
            if e.stdout:
                logger.info(e.stdout.decode())
            if e.stderr:
                logger.error(e.stderr.decode())
                logger.debug("=== STDERR ===")
                logger.debug(e.stderr)
    
>           raise RuntimeError(_s)
E           RuntimeError: '['gmake']' exited with an error: 2

edalize/edatool.py:483: RuntimeError
----------------------------------------------------------------------------------------- Captured stdout call ------------------------------------------------------------------------------------------
gmake[1]: Entering directory '/tmp/pytest-of-yuri/pytest-70/test_modelsim0/0'
gmake[1]: Leaving directory '/tmp/pytest-of-yuri/pytest-70/test_modelsim0/0'
```

yurivict, 2023-02-23:
Another error:
```
______________________________________________________________________________________________ test_oxide _______________________________________________________________________________________________

self = <edalize.oxide.Oxide object at 0x92381a460>, cmd = 'gmake', args = [], quiet = True

    def _run_tool(self, cmd, args=[], quiet=False):
        logger.debug("Running " + cmd)
        logger.debug("args  : " + " ".join(args))
    
        capture_output = quiet and not (self.verbose or self.stdout or self.stderr)
        try:
>           cp = run(
                [cmd] + args,
                cwd=self.work_root,
                stdin=subprocess.PIPE,
                stdout=self.stdout,
                stderr=self.stderr,
                capture_output=capture_output,
                check=True,
            )

edalize/edatool.py:460: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

input = None, capture_output = False, timeout = None, check = True, popenargs = (['gmake'],)
kwargs = {'cwd': '/tmp/pytest-of-yuri/pytest-72/test_oxide0/0', 'stderr': None, 'stdin': -1, 'stdout': None}, process = <Popen: returncode: 2 args: ['gmake']>, stdout = None, stderr = None, retcode = 2

    def run(*popenargs,
            input=None, capture_output=False, timeout=None, check=False, **kwargs):
        """Run command with arguments and return a CompletedProcess instance.
    
        The returned instance will have attributes args, returncode, stdout and
        stderr. By default, stdout and stderr are not captured, and those attributes
        will be None. Pass stdout=PIPE and/or stderr=PIPE in order to capture them.
    
        If check is True and the exit code was non-zero, it raises a
        CalledProcessError. The CalledProcessError object will have the return code
        in the returncode attribute, and output & stderr attributes if those streams
        were captured.
    
        If timeout is given, and the process takes too long, a TimeoutExpired
        exception will be raised.
    
        There is an optional argument "input", allowing you to
        pass bytes or a string to the subprocess's stdin.  If you use this argument
        you may not also use the Popen constructor's "stdin" argument, as
        it will be used internally.
    
        By default, all communication is in bytes, and therefore any "input" should
        be bytes, and the stdout and stderr will be bytes. If in text mode, any
        "input" should be a string, and stdout and stderr will be strings decoded
        according to locale encoding, or by "encoding" if set. Text mode is
        triggered by setting any of text, encoding, errors or universal_newlines.
    
        The other arguments are the same as for the Popen constructor.
        """
        if input is not None:
            if kwargs.get('stdin') is not None:
                raise ValueError('stdin and input arguments may not both be used.')
            kwargs['stdin'] = PIPE
    
        if capture_output:
            if kwargs.get('stdout') is not None or kwargs.get('stderr') is not None:
                raise ValueError('stdout and stderr arguments may not be used '
                                 'with capture_output.')
            kwargs['stdout'] = PIPE
            kwargs['stderr'] = PIPE
    
        with Popen(*popenargs, **kwargs) as process:
[... 447 lines cut ...]
                stdout=self.stdout,
                stderr=self.stderr,
                capture_output=capture_output,
                check=True,
            )
        except FileNotFoundError:
            _s = "Command '{}' not found. Make sure it is in $PATH".format(cmd)
            raise RuntimeError(_s)
        except subprocess.CalledProcessError as e:
            _s = "'{}' exited with an error: {}".format(e.cmd, e.returncode)
            logger.debug(_s)
    
            if e.stdout:
                logger.info(e.stdout.decode())
            if e.stderr:
                logger.error(e.stderr.decode())
                logger.debug("=== STDERR ===")
                logger.debug(e.stderr)
    
>           raise RuntimeError(_s)
E           RuntimeError: '['gmake']' exited with an error: 2

edalize/edatool.py:483: RuntimeError
----------------------------------------------------------------------------------------- Captured stdout call ------------------------------------------------------------------------------------------
```

olofk, 2023-02-23:
Thank you. This provide some additional clues. It looks like your tests are trying to run the actual commands, rather than the mock commands from https://github.com/olofk/edalize/tree/main/tests/mock_commands which are supposed to override the ordinary binaries.
I also notice that your tests run `gmake` instead of plain `make`. Is this a change you have made directly in the Edalize source or some clever alias? Given your FreeBSD affiliation I guess it's to use GNU Make over BSD Make. And in that case, I also wonder if there are some assumptions made about the path environment that doesn't hold true for BSD

yurivict, 2023-12-10:
0.5.3 has the similar failuses, see the [log](https://people.freebsd.org/~yuri/py-edalize-0.5.3-tests.log).

It appears that many failures are due to formatting differencres.
It isn't clear why formatting tests are a part of the testsuite. Sholudn't formatting be fixed at the development stage? Run-time results are not supposed to depend on code formatting.

gmake is the executable for GNU Make on BSDs. We apply these patches in the port:
```
        @${GREP} -rl '"make"' ${WRKSRC} | ${XARGS} ${REINPLACE_CMD} -i '' -e 's|"make"|"${GMAKE}"|'
        @${GREP} -rl '"python3"' ${WRKSRC} | ${XARGS} ${REINPLACE_CMD} -i '' -e 's|"python3"|"${PYTHON_CMD}"|'
        @${GREP} -rl 'shell python3 ' ${WRKSRC} | ${XARGS} ${REINPLACE_CMD} -i '' -e 's|shell python3 |shell ${PYTHON_CMD} |'
        @${GREP} -rl '$$(shell which python3)' ${WRKSRC} | ${XARGS} ${REINPLACE_CMD} -i '' -e 's|$$(shell which python3)|${PYTHON_CMD}|'
        @${REINPLACE_CMD} -i '' -e 's|/usr/bin/env python3|${PYTHON_CMD}|' ${WRKSRC}/tests/mock_commands/vcs
```

Ideally ```make``` and ```python3``` should be configurable at the build time.

olofk, 2023-12-14:
I'm also very confused about why it tries to do formatting as part of the test suite. I don't think I have seen that before. It's only supposed to run pytest. Manually running black locally, I also see a lot of errors, so we might as well just fix them. I created a freebsd branch where I have reformatted all files so that it's a bit easier to see what remaining issues we actually have. Is it possible for you to rerun the checks against that branch to see if we can get the number of errors down. There also seems to be a couple more errors and I'll try to look into those as well.

And point taken about make and python3. With the new flow API, it's a bit easier to change this centrally, so I'll look into doing that once we have the other stuff taken care of.