olofk/edalize #370: Skipping bitstream generation in vivado

javValverde, 2023-02-17:
Is there an option for running vivado without generating a bitstream?
This is specially useful when analyzing modules for resource utilization and timing (Hierarchical design UG905)

shareefj, 2023-02-21:
Isn't this a don't care?  The time spent generating a bitstream is so small that you won't notice it and it has no effect on being able to analyse the placement reports.

javValverde, 2023-02-22:
Not really, if I don't provide IO constraints (which I don't), the bitstream cannot be generated

shareefj, 2023-02-22:
I wouldn't have thought that you would get any sensible timing results without IO constraints but hey.

But going back to your original issue, why does the bitstream generation failing hurt you?  The reporting stage happens before the write_bitstream step.  So any reports will already have been written.  Edalize isn't in charge of the Tcl flow as under the hood it uses the project flow so there are no options to control this behaviour.

If for some other reason you really don't want the bitstream step run then you have a number of options:

1.  Run Fusesoc with the `--setup` flag only and then run `make build-gui` from the build directory. This will launch the GUI and you can do what you want.
2. Write your own Tcl hooks that write out whatever reports you want and then exit at whatever step you want to.

javValverde, 2023-02-27:
Thanks for the reply!

I'm analyzing whether I can use fusesoc/edalize as part of our continuous integration, so using a GUI is not an option.

I find it quite misleading to let the bitstream generation error, even though my intended use case worked.  It is specially hard to differentiate actual errors and the expected bitstream generation error.
I managed to execute a tcl hook, but am unable to gracefully stop vivado (This might be more related to vivado than to edalize)

shareefj, 2023-03-01:
What you could do is create a PR that updates the following template to enable different steps to be passed in.

https://github.com/olofk/edalize/blob/main/edalize/templates/vivado/vivado-run.tcl.j2#L12

javValverde, 2023-03-06:
I'm willing to make a PR, but I'm not really sure which is the best way to implement this.  
Maybe an approach similar to `--pnr None`? Something like `--skip-bitream`?

olofk, 2023-03-07:
I am currently refactoring the Vivado flow to increase the flexibility and avoid unnecessary rebuilds. Once that is done, it should be straight-forward to add an option to skip the bitstream step. If possible, it would be great if we could hold off adding a PR until then. I understand it's not ideal to wait for this, so I can help probably give you some pointers if you want to make a quick hack in the meantime

javValverde, 2023-03-07:
It would be really helpful if you could give me some hints on how to workaround this problem