olofk/edalize #374: filelist.py uses python 3.8 syntax

dbesse, 2023-03-28:
https://github.com/olofk/edalize/blob/8473fe13a1abffe013fae8d7716bba856d770efe/edalize/filelist.py#L135

The "walrus operator" is new in python 3.8 (see https://docs.python.org/3/whatsnew/3.8.html#assignment-expressions). This is seen by CI failing in 3.7 (https://github.com/olofk/edalize/actions/runs/4515708053/jobs/7953312081).

shareefj, 2023-03-28:
That code will likely disappear as part of #372

dbesse, 2023-03-28:
So, can it be removed from the release? The syntax error in this file prevents edalize version 0.5.0 from being used with python3.7, even if that tool/flow is not used.

shareefj, 2023-03-28:
I'd suggest moving to a previous version unless @olofk can do a quick patch release.

alperyazar, 2023-03-30:
Came here to open the same issue. One of our old CI system runs with Python 3.6. Although the latest 0.5.0 version is marked as compatible with Python >= 3.6, it is not. The latest version requires at least 3.8. Since the CI system of this repo looks testing at least version 3.7 (which is failing), I think that [setup.py](https://github.com/olofk/edalize/blob/main/setup.py#L89) needs an update too, requiring at least 3.7 since CI is not testing 3.6.

olofk, 2023-05-17:
The filelist backend has now been temporarily removed and Edalize 0.5.1 is now released which should hopefully fix this.

closed 2023-05-17 by olofk