olofk/edalize #376: Why is the user guide html that I generated empty.

salmansheikh, 2023-04-01:
[image]
I installed all the tools for make html work and even tried make singlehtml and there seems to be nothing but these titles with no links for the user guide.

imphil, 2023-04-02:
Thanks for your bug report, @salmansheikh! Unfortunately, I'm not quite sure what your problem is. Can you show the commands you ran, and what you expected? Your screenshot looks very much like https://edalize.readthedocs.io/en/latest/user/index.html, i.e. the empty Configure/Build/Run sections are expected.

salmansheikh, 2023-04-02:
Ok.

closed 2023-04-04 by imphil

salmansheikh, 2023-04-04:
Someone else wrote this for putting RISCV onto an Upduino with edalize but
it fails. I did the install of edalize correctly afaik. Attached is the
file.

***@***.***:~/risky_bizness/grv/designs/upduino3$ python3
upduino3.py
Init the backend
Traceback (most recent call last):
  File "upduino3.py", line 87, in <module>
    backend = get_edatool(tool)(edam=edam, work_root=work_root)
NameError: name 'get_edatool' is not defined