olofk/edalize #378: Efinix FPGA support

sebinho, 2023-04-06:
Hi Guys,

Is there any Efinix FPGA support that is planned in the future?

Thanks!

olofk, 2023-05-17:
Would love to have that and hope that someone will add it.

RiceShelley, 2023-10-03:
Is there an active branch / anyone working on this? If not I'd like to try adding it.

sebinho, 2023-10-04:
I made my own implementation that is working well for my use case. I would be willing to share it but knowing that it most likely is not yet production ready and might have to be tuned in order to be put on the main branch.

RiceShelley, 2023-10-04:
I would love to take a look / help streamline.

sebinho, 2023-10-06:
I created a pull request that includes my code:
https://github.com/olofk/edalize/pull/395

My fork can be found here:
https://github.com/sebinho/edalize

If you have any questions on how to use it, let me know.
Cheers

olofk, 2023-10-06:
Thanks for this! I will look at the PR and add comments there.

Also, let the Efinity toolchain serve as an example why you should never ever ask your EDA tool vendors to supply a Python interface for their tools :)

olofk, 2023-10-10:
Well then. Happy to report we now have an Efinity backend. Thanks for that @sebinho. Will push a Servant port, LED to Believe port and CoreScore as well for my Xyloni board to have some examples

closed 2023-10-10 by olofk