olofk/edalize #379: ImportError: cannot import name 'get_edatool' from 'edalize' (unknown location)

rafaelnp, 2023-04-06:
I started having this error when using edalize  0.5.0, insalling it using pip:

```sh
 pip install --user edalize
```

Running a minimal project I get the following error:
```sh
❯ python min.py
Traceback (most recent call last):
  File "/home/user/src/min/min.py", line 2, in <module>
    from edalize import get_edatool
ImportError: cannot import name 'get_edatool' from 'edalize' (unknown location)
```

when I run the Arch Linux provided package (edalize 0.4.0), the error does not show up.

```sh
❯ python min.py
vivado -notrace -mode batch -source min.tcl

****** Vivado v2022.2 (64-bit)
  **** SW Build 3671981 on Fri Oct 14 04:59:54 MDT 2022
  **** IP Build 3669848 on Fri Oct 14 08:30:02 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source min.tcl -notrace
create_project: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 1321.230 ; gain = 0.023 ; free physical = 2526 ; free virtual = 33805
INFO: [Common 17-206] Exiting Vivado at Thu Apr  6 20:59:59 2023...
vivado -notrace -mode batch -source min_run.tcl min.xpr
...
...
```

Infos:
* Linux Kernel: 6.2.6-arch1-1
* Python: 3.10
* pip 23.0.1
* Vivado 2022.2

Did I forget to install anything or any subtle error?  Can you reproduce the error?

the minimal project has the following files:

min.py:
```py
from edalize import get_edatool
import os

# build directory
work_root = 'build'

file_list = [
    {'name': os.path.relpath('min.vhd', work_root), 'file_type': 'vhdlSource-2008'},
    {'name': os.path.relpath('min.xdc', work_root), 'file_type': 'xdc'},
]

tool = 'vivado'
edam = {
    'files': file_list,
    'name': 'min',
    'parameters': {},
    'toplevel': 'min',
    'tool_options': {
        'vivado': {
            'part': 'xc7a35tcpg236-1'
        }
    }
}

# Get the vivado backend
backend = get_edatool(tool)(edam=edam, work_root=work_root)
backend.configure()
backend.build()
```

min.vhd:
```
library ieee;
context ieee.ieee_std_context;

entity min is
        port (
                clk : in  std_logic;
                val : out std_logic
        );
end entity min;

architecture rtl of min is
        signal int_val : std_logic := '0';
begin
        val <= int_val;

        main: process(clk)
        begin
                if rising_edge(clk) then
                        int_val <= not int_val;
                end if;
        end process main;
end architecture rtl;
```

min.xdc:
```xdc
set_property -dict { PACKAGE_PIN W5 IOSTANDARD LVCMOS33 } [get_ports { clk }]; #IO_L12P_T1_MRCC_34 Sch=CLK100MHZ
create_clock -add -name sys_clk_pin -period 10.000 -waveform {0 5.000}  [get_ports { clk }];

## LED
set_property -dict { PACKAGE_PIN U16 IOSTANDARD LVCMOS33 } [get_ports { val }]; #IO_L23N_T3_A02_D18_14 Sch=led[0]

## Configuration options, can be used for all designs
set_property CONFIG_VOLTAGE 3.3 [current_design]
set_property CFGBVS VCCO [current_design]
```

olofk, 2023-04-12:
Yes. This is unfortunately a backwards-incompatible change I had to make to support external plugins. The function still exists but have moved, so change the import line to `from edalize.edatool import get_edatool` and it should work again

rafaelnp, 2023-04-14:
Thanks for your reply, it did solve the problem.

closed 2023-04-14 by rafaelnp

TheZoq2, 2023-10-11:
The README still references get_edatool as imported from `from edalize import*`

olofk, 2023-10-12:
You're right. This should be fixed