olofk/edalize #399: Simulating Vivado IPs (xci, bd)

adrianmrd, 2023-10-13:
Hi,

I'm using fusesoc+edalize for synthesis on zynq mpsoc platform. 
Now I would like to also simulate some of my subdesigns with xsim (or any other simulator for that matter), which is not a straightforward task when the subdesigns are described with xci/bd files. Is there any solution planned for this issue?

I have been looking into the new flow api (sim flow in particular) and I think it should be possible to add vivado as a backend for generating the simulation sources and passing them on to xsim (or other simulation tool).
I have the bandwidth for implementing such a feature and would be more than happy to contribute, but I'm not sure if it's the right approach. 
Any thoughts?

fpgauserdude, 2023-11-21:
I am also interested in this feature. This is the biggest drawback of Edalize, in my opinion.

olofk, 2023-11-24:
I have been thinking about this issue since almost the start of FuseSoC, so I have plenty of ideas but not yet an implementation :) There are a couple of different ways to solve this. In my mind, the most generic solution is to create a generator that gets an xci file (or even a tcl script?) as input, runs Vivado to create the simulation files+ipxact description, parses the ipxact description to learn what files that were created and adds them to the generated core. This way would work with any simulator. As you mention, it could be solved with a frontend as well if we want to keep it on the Edalize side. It's mostly that I see it as something that operates on a core, rather than a whole system. It also allows easier caching of the generated results.

For xsim specifically, I think it would make sense to use xsim in project mode (like the vivado tool class does) because I believe this would allow us to add xci files directly and Vivado would know what to do with them. It would probably give some other benefits as well by reusing some convenience features already implemented in the vivado tool class.

olofk, 2023-11-24:
Forgot to mention! I did a proof of concept doing pretty much this, but for Xilinx ISE, where I wrote a generator that got cgp files as input, ran coregen, parsed the ipxact output and generated a core with the created simulation files. Never published it, but might be able to find it somewhere if you want some inspiration

adrianmrd, 2023-11-27:
I tried playing around with the flow API, having Vivado as a frontend to generate simulation sources, but this resulted in a tighter-than-I-would-like coupling between vivado and the backend tool. Also a weird execution flow in edalize from python (project setup; generate tcl scripts) -> vivado (generate sources) -> python (update edam and generate simulation Makefile). 

Anyway I agree that generator scripts are an all-around better solution for this issue (not least for preserving backwards compatibility). I would very much appreciate your proof of concept work to use as a starting point if you can find it.