olofk/edalize #403: `make_libraries_directories` on Windows fails due to `mkdir -p`

pintert3, 2023-10-24:
For windows CMD and Powershell, the `-p` is not necessary for the mkdir command, and is unrecognized. A work around I'd suggest is to check if the OS is Windows, and remove the `-p` in `edalize/ghdl.py`

pintert3, 2023-10-27:
I got a temporary fix for windows users, which I've discussed in https://github.com/TerosTechnology/vscode-terosHDL/issues/472#issuecomment-1782866090

Will need to just do an OS check to offer an alternative command.

creiter64, 2023-11-06:
I'm also hit by this, it seems edalize 0.5.1 didn't have this problem.

olofk, 2024-01-25:
Wrote a response in https://github.com/TerosTechnology/vscode-terosHDL/issues/472 but forgot to write here. Do we need `-p` at all?

pintert3, 2024-01-29:
I must have missed the notification in the other issue. I haven't really over scrutinized the exact intention of the operation in the code, but from the looks of things, we're simply making library directories based on the names of the libraries. According to my knowledge, the `-p` would only be useful if we were likely to overwrite an existing directory. I don't know if we plan on using the same folders every time a run is triggered(haven't scrutinized enough), but if so, then it might be necessary. Otherwise, we don't need it at all cause every library seems to have it's own folder.